supermerill/SuperSlicer #1884: Auto Speed weird behaving

ocgeek, 2021-11-15:
### Version
SS 3.6.0.9

### Operating system type + version
Windows 10

### Behavior
* AutoSpeed generated feed rates do not match the settings
* One Step - Slice and inspect the resulting G-Code
* Different feed rates as per Auto Speed settings
* Feed rate appears incorrect

I'm using 0 infill speed so to use the Auto Speed setting
I have a Dragon so i set the Volumetric to 15 mm/s^3 (1.75 mm filament / 0.4 mm nozzle)

Case 1)
I set max print rate to 100 mm/s
In the generated g code the resulting feed for _Internal infill_ (mm/min) is ~2971 which is 45 mm/sec.... weird

[image]

Case 2) 
I set the max print rate to 300 mm/s (well above the max 15mm/s^3 can do)
in the generated g code the resulting feed appears higher than the one which you calculate given the line width, nozzle dia, ...

[image]

With those settings it shall be 104 but it generates 110
[image]

thanks

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs

[Auto Speed Test.zip](https://github.com/supermerill/SuperSlicer/files/7537276/Auto.Speed.Test.zip)

csch0, 2021-11-15:
I think the first one should be fixed in the latest release just give it a try (https://github.com/supermerill/SuperSlicer/releases/tag/2.3.57.5)

ocgeek, 2021-11-15:
I confirm the first is fixed in 2.3.57.5

This is gcode out of setting max auto 100 mm/sec

[image]

supermerill, 2021-11-16:
Did you use spacing for your calculus?
To have the flow per mm, you can multiply the spacing by the height.
If you use the width, you have to remove the bit around the half-circles.
see https://manual.slic3r.org/advanced/flow-math

closed 2021-11-19 by supermerill