supermerill/SuperSlicer #1886: FR - Add Total Layers to the Gcode metadata

vajonam, 2021-11-15:
Right now there is a mismatch when supports are used, there is no way for the klipper/moonraker to determine the correct number of layers when using variable layer height or supports which introduces variable layer height.  

**Describe the solution you'd like**
Can we add `total_layer_count` to the medata printed. I know I can add it custom G-code but this is for moonraker / klipper that allows it correctly reflect the total layers. .

**Describe how it would work**
Just an additional parameter at the end gcode along with all the other info. 

**Describe alternatives you've considered**
adding it to custom G-code.

vajonam, 2021-11-17:
@supermerill Thanks for the quick response! Do I close this? or will you close it w/ next version?

supermerill, 2021-11-17:
you can close when you verify that it works as expected.

vajonam, 2021-11-22:
Confirmed this is work. Will close the issue. Thanks

closed 2021-11-23 by vajonam