supermerill/SuperSlicer #1887: Internal bridge infill has wrong layer height

blex-max, 2021-11-15:
### Version
2.3.57.5

### Operating system type + version
Ubuntu 20.04 LTS

### Behavior
Even though I have disabled variable layer heights in settings, internal bridge infill prints with a greater layer height than the rest of the model. No other feature types are affected. I am expecting all features to have the same layer height. I am unable to find any setting which would be causing this. It seems to occur on internal bridge infill regardless of model sliced.

[image]
_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[m12_attachment_variable_layer_height_issue.zip](https://github.com/supermerill/SuperSlicer/files/7540046/m12_attachment_variable_layer_height_issue.zip)

neophyl, 2021-11-16:
Technically it doesn't.  This is the way Slic3r/Prusa Slicer and Super Slicer all work (at the moment).  
Any extrusion classed as bridging (so bridging perimeters, bridging infill, layers above a raft etc, because it isn't being squished into the layer below the slicer models it as having the same width as your nozzle.  Its always worked this way.

Its an annoyance yes.  Its why I often used to turn detect bridging off in PS or set the thresholds for bridging flow in SuSi to a really high amount.  Unfortunately there's no setting to change the bridge infill though.  As that is inside the model in practice its not a problem.

I mention at the moment though as Prusa Slicer 2.4 which is currently in beta has an option called thick bridges, when disabled it changes that underlying logic (which is buried deep into the codebase).  As such when PS 2.4 has a proper release and is eventually merged into Super Slicer then the same option **should** make it over.  

Until then this is intended operation and not a bug.

blex-max, 2021-11-16:
So it is not trying to print with a variable layer height? If this is the case then I'll close this issue. Thanks!

supermerill, 2021-11-16:
there is nothing under the extrusion to flatten it, so to have the required width (nozzle), you must extrude much more plastic, hence the big height.

closed 2021-11-16 by supermerill