supermerill/SuperSlicer #1888: ETA Print time inaccurate in version 2.3.56

TWIISTED-STUDIOS, 2021-11-15:
**Describe the bug**
When I slice a file on the super slicer application the estimated print time comes out at about 9hrs, 45mins, when printing said file it prints in about 1hour 50ish give or take, using the g-code analyzer website the estimated time is 1:39:53 im unsure where the slicer is getting an extra 8 hours roughly on the print time. 

**SlicedGcode**
[Cutter_-_Christmas_Tree.zip](https://github.com/supermerill/SuperSlicer/files/7541376/Cutter_-_Christmas_Tree.zip) or was there some other format that was needed to be exported?

**Expected behavior**
Estimated print time to be accurate. 

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version Version 2.3.56.9

neophyl, 2021-11-16:
It’s better to save your project from super slicer and then zip up and attach the 3mf file. The project will contain all the settings as well as the model. Makes it much easier to debug things.

supermerill, 2021-11-16:
it use a virtual marlin firmware to estimate the slicing time. The settign of this virtual firmware are in the printer-> printer limit.
Please be sure they are as similar as possible with your printer.
When it's set, you can use the `time_estimation_compensation` field to correct for firmware variation if you're not using marlin.

TWIISTED-STUDIOS, 2021-11-16:
thank you for the fast and quick responces I honestly do not know what is going on, I just opened the application back up to save the project and it seems the time is back to a reasonable time again, I will upload the zip file anyway. 
[Cutter_-_Christmas_Tree.zip](https://github.com/supermerill/SuperSlicer/files/7547514/Cutter_-_Christmas_Tree.zip)

supermerill, 2021-11-16:
There were some occurrence when the time estimation bugs out to a 2 Millard +days, but it's a very rare and unreproducible bug.

TWIISTED-STUDIOS, 2021-11-16:
Seems like it could have been a random bug then?