supermerill/SuperSlicer #1890: Odd behaviours with support materials and toolchanging printers

tdcox, 2021-11-15:
**Describe the bug**
Extruders / Tools are allocated incorrectly when slicing.

**To Reproduce**
Steps to reproduce the behavior:
Set up system to support Toolchanging printer with 4 x single extruder tools and 1 x milling tool.
Create and allocate a different material for each tool.
Create a single standard calibration cube and initially leave it set to the default material.
Enable **Print Settings/Support material/Generate support material** and **Print Settings/Support material/Auto generated supports**.

Test by settng the following combinations of settings under Print Settings/Multiple extruders/Extruders and then slicing, as per the tables below:

(For conciseness, the settings 'perimeter extruder, infill extruder and solid infill extruder are referred to as 'Print' and the settings support material extruder and support material/raft interface extruder are referred to as 'Support'. All values in each group are set to the values shown below for 'Print' and 'Support' respectively)

| Settings | Selected | Rendered |
| -------- | -------- | ---------- | 
| Print      | 1            | T0           |
| Support | 1            | T0           |

| Settings | Selected | Rendered |
| -------- | -------- | ---------- | 
| Print      | 2            | T1           |
| Support | 2            | T1           |

| Settings | Selected | Rendered |
| -------- | -------- | ---------- | 
| Print      | 3            | T2           |
| Support | 3            | T2           |

| Settings | Selected | Rendered |
| -------- | -------- | ---------- | 
| Print      | 4            | T3           |
| Support | 4            | T3           |

| Settings | Selected | Rendered |
| -------- | -------- | ---------- | 
| Print      | 1            | T0           |
| Support | 0            | T0           |

| Settings | Selected | Rendered |
| -------- | -------- | ---------- | 
| Print      | 1            | T1           |
| Support | 2            | T0           |

| Settings | Selected | Rendered |
| -------- | -------- | ---------- | 
| Print      | 1            | T2           |
| Support | 3            | T0           |

| Settings | Selected | Rendered |
| -------- | -------- | ---------- | 
| Print      | 1            | T3           |
| Support | 4            | T0           |

| Settings | Selected | Rendered |
| -------- | -------- | ---------- | 
| Print      | 4            | T0           |
| Support | 1            | T3           |

Initially, the results seem as expected for the matching materials and the 'use current extruder' combos, however as soon as we switch to different materials for object and support, _it is clear that the Support tool is being used for the print and the Print tool is being used for the supports._

Now, in the Plater view, set the extruder for the calibration cube object to extruder 2 / T1 in the right hand pane and run the tests again:

| Settings | Selected | Rendered |
| -------- | -------- | ---------- | 
| Print      | 1            | T0           |
| Support | 1            | T1           |

| Settings | Selected | Rendered |
| -------- | -------- | ---------- | 
[... 2178 lines cut ...]
tool_name = T0;T1;T2;T3
toolchange_gcode = T[current_extruder]
use_firmware_retraction = 1
use_relative_e_distances = 1
use_volumetric_e = 1
variable_layer_height = 1
wipe = 0,0,0,0
wipe_advanced = 0
wipe_advanced_algo = linear
wipe_advanced_multiplier = 60
wipe_advanced_nozzle_melted_volume = 120
wipe_extra_perimeter = 0,0,0,0
wipe_speed = 0,0,0,0
z_offset = 0
z_step = 0.000625

[presets]
print = E3D Toolchanger - Supports
sla_print = 
sla_material = 
printer = E3D Toolchanger
filament = KeHuaShiNa Blue Silk PLA
filament_1 = Esun ePA-CF
filament_2 = SpoolWorks Scaffold
filament_3 = SpoolWorks FleXD

neophyl, 2021-11-16:
You do realise that if you saved your project file as requested the resulting 3mf file would have ALL the settings you listed as a massive list include automatically and would act as a snapshot of your settings making it much easier for the developer to try and recreate the problem.

tdcox, 2021-11-16:
Due to the number and severity of the issues with the product currently, I have no reason to assume that a simple project file contains sufficient information, so provided a full export to support recreating the complete environment if necessary.

The report shows that there is nothing in the project file that wasn't created by SuperSlicer itself, so no need to include object information with this issue.

You do realise that behaving passive-aggressively towards people reporting issues on open source projects nearly always causes them to lose interest in contributing to them further? A polite request for additional information, if needed, is usually much more productive in building a relationship.

neophyl, 2021-11-16:
Sorry I think we have differing definitions of passive aggressive.
I was merely trying to point out that the form for opening an issue does request a saved project file when opening an issue

To Quote -
**>> Project File <<**
*Please* upload a ZIP archive containing the project file used when the problem arise. Please export it just before the problem occur. Even if did nothing and there is no object, export it as it contains your current configuration.

A project file contains all those settings and is the main reason why such a request is made.  A simple cube included on the plater if you cant include the actual object is fine, its more your saved profiles and how they are configured that are important as long as that shows the same issues.  

Now if you want an example of passive aggressive, I could point out that what you assume or don't assume has no bearing on the requested information.  See that was properly aggressive.  Merely as an example.  
Please don't take everything typed in an online context as a personal attack.  Its not generally.  My earlier response was not meant as an attack on you, if you took it that way then I apologise.  It was 6am where I am and maybe I could have worded it better.

tdcox, 2021-11-16:
[SwappedMaterials.3mf.zip](https://github.com/supermerill/SuperSlicer/files/7546662/SwappedMaterials.3mf.zip)

Here is a project file for one of the scenarios. Please let me know if you need me to upload the other 17 I tested as well?

supermerill, 2021-11-16:
you have 
`T[current_extruder]` in your custom toolchange gcode instead of `T[next_extruder]`, look at the tooltip.

tdcox, 2021-11-16:
Ah, interesting! I inherited that from the official E3D SuperSlicer profile so I will raise a bug against that, thanks.

I don't have any tooltips on the Mac? Is that a PC only feature?

supermerill, 2021-11-16:
> I don't have any tooltips on the Mac? Is that a PC only feature?

you should. But the macos has very short tooltip time and they can't be changed.
Go to configuration -> preference -> GUI -> appearance -> check/uncheck custom tooltip, use the one that works best.

tdcox, 2021-11-16:
Ah, found out what is going on. In fullscreen mode, all pop-up panes including tool tips, selection drop-downs and search were appearing on the wrong virtual desktop. So, if SuperSlicer is running maximised as Desktop 6, pop-ups would appear invisibly on Desktop 5. This was only apparent when I clicked on a stateful pop-up (filament selector) and nothing appeared to happen, but when I swiped to another desktop, the selector was sitting there on its own.