supermerill/SuperSlicer #1891: Coordinate outside allowed range

wesk-ms, 2021-11-16:
New bug for same issue reported recently in issue #66

**Describe the bug**

Import any STL file
Move the part to a different part of the bed (move it enough that the new location isn't overlapping where it was originally).
Configure the part to be printed with a brim.
Slice as usual.

During the slicing operation, I receive a dialog box that says "Coordinate outside allowed range"

**>> Project File <<**

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: [e.g. MacOS, Linux, Windows]
 - Version [e.g. 2.3.55.5], Use `Help->About`_

**Additional context**
Add any other context about the problem here.

supermerill, 2021-11-16:
seems like a duplicate of #66

wesk-ms, 2021-11-16:
Issue #66  is very old and seems to include multiple issues that are similar in nature.  I agree that this issue is a duplicate of the most recent problem reported in #66.  However, some of the earlier problems mentioned in #66 don't seem to be related to Brim usage.

I created a new issue to specifically track this error relative to the Brim feature.

closed 2021-12-15 by supermerill