supermerill/SuperSlicer #1892: I need to know if there is a placeholder for the total number of layers.

Denniswie, 2021-11-16:
I need to know if there is a placeholder for the total number of layers.
I have checked the Slic3r manual, (  https://manual.slic3r.org/advanced/placeholder-parser  ) and there seems to be only
 "layer_num" - current active layer.
I would like to modify on layer change g-code to be:
M117 Layer [layer_num] of [total_layers] layers
So I guess my question is - Is the list of placeholders in the manual all there is ?

_Originally posted by @Denniswie in https://github.com/supermerill/SuperSlicer/discussions/1797#discussioncomment-1647282_

supermerill, 2021-11-16:
`total_layer_count`
see https://github.com/supermerill/SuperSlicer/wiki/Variable-list

Denniswie, 2021-11-16:
Many many thanks ( also for the link to wiki - should have gone there first )
Lots to keep me quiet there . :-)

closed 2021-11-16 by Denniswie