supermerill/SuperSlicer #1893: Crash when generating infill

pushc6, 2021-11-16:
**Describe the bug**
When slicing a particular plate SuperSlicer will crash when generating the infill

**To Reproduce**
Steps to reproduce the behavior:
Slice the attached 3mf

**>> Project File <<**
Attached

**Expected behavior**
For it not to make me sad by blowing up

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: MacOS
 - Version 2.3.57 but also happened in 2.3.56
[crash_ss.3mf.zip](https://github.com/supermerill/SuperSlicer/files/7546709/crash_ss.3mf.zip)

**Additional context**
Add any other context about the problem here.

supermerill, 2021-11-16:
That's a bug in the monotonous code from prusa.
I don't really know how it works.
Will see if I can create a dirty fixing patch.
ideally, I should create a unit test that show the problem and give that to prusa for debugging.

pushc6, 2021-11-16:
Thanks for the quick response @supermerill, I was able to get around it by just spinning print 90 degrees. So for anyone else out there that may encounter this, it seems spinning seems to help.

closed 2021-12-15 by supermerill