supermerill/SuperSlicer #1898: Adding placeholder for current_pos_x &  current_pos_y for custom gcode

DzzD, 2021-11-16:
I have now fully switched to Super Slicer, goodbye Simplify 3D !, but there is something I have trouble to find and dont know how and where I should do this.

I am looking for a purge bucket script, so that each time the extruder need to change color it will extract 10cm of filament at the left most of the bed into a bucket.

This way no need of purge block, printing is a lot faster and less filament is lost.

It was working very well with Simplify 3D, but did not find script already made or documentation to how to do it in Super Slicer.

Any help would be grealty appreciated.

_Originally posted by @DzzD in https://github.com/supermerill/SuperSlicer/discussions/1855_

supermerill, 2021-11-16:
can I add current_pos as an array of two double?
so current_pos[0] for x and current_pos[1] for y

like for relative_correction, print_bed_size, ...

closed 2021-12-15 by supermerill

DzzD, 2021-12-17:
Just to be sure, what is exactly current_pos, the last pos used or the next pos ?

I am wondering if next pos would not be a more interesting data, I mean after a custom gcode you probably want to go to the next pos rather than go back to the last one, no ?

supermerill, 2021-12-17:
It's the position of the nozzle at the start of the custom gcode section.