supermerill/SuperSlicer #1902: Place on face freezes for round items

greembow, 2021-11-16:
**Is your feature request related to a problem? Please describe.**
When using place on face on a model that is either very round or has many vertices, the program will freeze until the calculation finishes.

**Describe the solution you'd like**
Allow SuperSlicer to continue running while it preforms the place on face calculations.

**Describe how it would work**
I'm not exactly sure how it would work, but you could probably use multithreading to push the place on face calculations to another thread so SuperSlicer will continue running. You could add a progress bar somewhere that pops up showing you the time left as well.

**Describe alternatives you've considered**
Just wait. This isn't a big deal at all, just something that would make using the slicer easier and more enjoyable.

**Additional context**
None to give. Just wanted to give a thanks for the SuperSlicer team for making a phenomenal slicer!

supermerill, 2021-11-16:
project file?

greembow, 2021-11-16:
[Visor.zip](https://github.com/supermerill/SuperSlicer/files/7549986/Visor.zip)

Sorry about that, I didn't think one was required. Here is a project where it takes a while for it to load. I had to zip it first because github won't let me upload .3mf files.

supermerill, 2021-11-16:
It takes me 2 seconds to compute the faces. There is no real freeze.
How long does it freeze you?

neophyl, 2021-11-16:
Couple of seconds for it to display the possible face areas for me too.  That's with 2.3.57.5 (and also with PS2.4 beta as I happened to have it open too).

greembow, 2021-11-17:
I'm running on Arch Linux, and it takes me about 5 seconds. Granted I don't have the fastest CPU. I just put in an issue because the first time it happened I thought the program had crashed. This isn't the only model that does it, I can find some others with longer freeze times if you want.

supermerill, 2021-11-17:
I can see how it may be a bit long on laptop cpus (or raspberry pi).
I won't work on it. Not the time nor the will to spend the days to decouple that from the GUI thread.
Maybe PrusaSlicer is more willing to fix this problem.