supermerill/SuperSlicer #1906: Error on opening .stl file

saenglert, 2021-11-17:
**Describe the bug**
When trying to open the cube.stl used in Teaching Techs calibration tutorial, SuperSlicer reports the error: `ERROR: First layer height can't be greater than max laye height` and the slice button says `Invalid data".

**To Reproduce**
Steps to reproduce the behavior:
1. Download the [file](https://teachingtechyt.github.io/files/cube.stl)
2. Open in SuperSlicer
4. See error

**Project File**
See file above

**Expected behavior**
SuperSlice should be able to open and slice the file without error.

**Desktop**
 - OS: Ubuntu LTS 20
 - Version 2.3.57.5

**Additional context**
Using the same .stl in Cura works without any problems.

The only relating issue I could find was #1474, but the layer heights in my settings are not set to zero:
[image]

Any help is greatly appreciated.

supermerill, 2021-11-17:
Next time, save your project file (File -> save as). The stl is useless for me. And the screen shot doesn't tell the whole story.

You have your first layer height higher than 0.32mm. That's why he refuse to slice, as an extrusion can't be taller than wider.
It's in print settings -> slicing -> first layer height.

saenglert, 2021-11-17:
Oh, ok. I thought you could start from the .stl file. Sorry then.

You are right, though. The setting for the first layer height is incorrect. Thank you very much for the help!

closed 2021-11-17 by saenglert