supermerill/SuperSlicer #1911: 2.3.57.5 impossible floating bridge/overhang perimeter

d-schmidt, 2021-11-17:
**Describe the bug**
There is a one layer bridge in this model. The slicer starts with the dark blue overhang perimeter in midair.
[image]
layer 95:
[image]
layer 96:
[image]
layer 97:
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. download Cam.stl https://www.thingiverse.com/thing:4945227/files
2. slice at 0.2 layer height

**>> Project File <<**
[btn_Cam.zip](https://github.com/supermerill/SuperSlicer/files/7558032/btn_Cam.zip)

**Expected behavior**
I expect one big bridge.

**Desktop (please complete the following information):**
 - win64

supermerill, 2021-11-18:
remove one perimeter on top, at least on this layer.

duplicate of #1867

d-schmidt, 2021-11-18:
weird interaction but okay. I guess it sees the middle circle as top?!

closed 2021-11-18 by d-schmidt