supermerill/SuperSlicer #1913: Crashes: Bridge Flow Ratio Calibration

HarryDevSV, 2021-11-18:
_Version of SuperSlicer used goes here (help->about)_
2.3.57.5

_What OS are you using, and state any version #s_
Windows 10 64bit

### Behavior
* _Describe the problem_
Bridge Flow Ratio Calibration fails to generate g-code after pressing "Test Flow Ratio" button  - Crashes App

_Is this a new feature request?_
No

[image]
[image]

tome9111991, 2021-11-18:
Same like https://github.com/supermerill/SuperSlicer/issues/1877 ?

You can test nigthly version
See "Action" menu on github

adoublard, 2021-11-18:
Same problem.
Windows 11 64bits

nigthly version work fine

LeonKnook, 2021-11-18:
nightly version win10 64 works fine!
Thank you very much!!