supermerill/SuperSlicer #1914: Cannot make large bed size like X=250mm Y=1550mm Z=400mm

LeonKnook, 2021-11-18:
Ya, I have a CR10 conversion to 1550mm long.
Cannot make large bed size like X=250mm Y=1550mm Z=400mm
I wanted to make such a custom profile but SuperSlicer doesn't like that. The application crashes.
I don't know if this is a bug or maybe a Pull request.
Prusa slicer doesn't want that either.

Any help is much appreciated

kind regards,
Leon

LeonKnook, 2021-11-18:
[image]
I tried it on another Windows 10 machine and all of a sudden there are no more problems. I have now completely finished the wizard and indicated to adjust the settings, maybe this is the solution.

neophyl, 2021-11-18:
Prusa slicer has a maximum bed size allowed.  Its a known limitation due to some of the math routines and number size.  

I seem to recall Merill making some changes to allow larger beds to be defined but I cant remember what form that took.

supermerill, 2021-11-18:
I removed some limitations.
I think there may be problems with rectilinear infill, as some code assumes it can store a square of a position in a int64. But that should be it.
If you have such a printer, please report any reproducible weird behavior.
If the error is random, try to limit the slicer to only one CPU core (via the task manager).

LeonKnook, 2021-11-18:
Thanks for the advice. I have to work now with this profile and see if I get stuck on problems.