supermerill/SuperSlicer #1916: request speed change

dave57000, 2021-11-18:
Hi love what you are doing 
just a quick request at least I hope it's quick, can you add a global speed change to the "height range modifier" so we can change the speed that the printer prints at different rights I have been using this on cura (to print ABS) to slow down the first few layers to improve adhesion and reduce warp with great success 
many thanks 
stay safe 
### Version
_Version of SuperSlicer used goes here (help->about)_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
* _Expected Results_
* _Actual Results_
  * _Screenshots if useful_

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_

supermerill, 2021-11-18:
kind of duplicate of #1750, right?

neophyl, 2021-11-18:
As a workaround if all you want is a general slowdown you could use an M200 S with a suitable percentage (same as dialling down the speed from a printers control panel) and then using the insert custom gcode at layer function in the slicer use M200 S100 to put it back to full speed at whatever layer you want.  Assuming your printer firmware supports that gcode naturally. 

You could insert several at differing layers so that it sped up as it gets higher.

dave57000, 2021-11-18:
thanks so much, I'll try that
keep safe

dave57000, 2021-11-18:
hi, #1750 looks like a lot more that I don't really understand. I was thinking of changing speed a layer or height ie at layer 5 increase speed by 300% or to "x" speed then you can start slow and then speed up when it's stuck. in cura, I start at 50mm/sec and ramp it up by 300% with a post-processing add on and it works fantastic with ABS
keep safe