supermerill/SuperSlicer #1917: Mid-air perimeter printed where it's not needed

tetele, 2021-11-18:
**Describe the bug**
I have a suspended surface that has a hole in it. Since I wanted to avoid supports, I "filled" the first 0.2mm (=layer height) of the hole in hope of a bridge from side to side.

However, Superslicer tries to draw the hole perimeters (even though there is no hole in that particular layer, and it does so by printing them first, in mid-air.

**To Reproduce**
Steps to reproduce the behavior:
1. Open attached project and slice it
2. Notice how on layer 16 the hole perimeters are printed mid-air

**>> Project File <<**
[ss_support.zip](https://github.com/supermerill/SuperSlicer/files/7563901/ss_support.zip)

**Expected behavior**
Not sure what should happen in general with these hole-type features. But in this case, since the layer is actually full, I would have expected to have continuous bridge lines from side to side.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version: 2.3.57.5

supermerill, 2021-11-18:
duplicate of #1911 #1867

supermerill, 2021-11-18:
note that you can select "no perimeter on bridge area " -> "fill the void with bridges" to have the same result without the stl hack.

tetele, 2021-11-18:
> duplicate of #1911 #1867

Not sure how #1867 is a duplicate, but sorry for not seeing #1911.

> note that you can select "no perimeter on bridge area " -> "fill the void with bridges" to have the same result without the stl trick.

Glad you mentioned that, as that option doesn't seem to do anything in any of my projects. Is there any documentation on how to use it apart from the tooltip?

tetele, 2021-11-18:
I just assumed that documentation was out of date, as it doesn;t work as advertised. See this test project [ss_fillvoid.zip](https://github.com/supermerill/SuperSlicer/files/7564405/ss_fillvoid.zip)

[image]

supermerill, 2021-11-18:
weird, it works on my test piece, but not on your real-world one.
[image]
I'll investigate

tetele, 2021-11-18:
Here's the STL if you need it for testing.

[DELETEME test.zip](https://github.com/supermerill/SuperSlicer/files/7565403/DELETEME.test.zip)

supermerill, 2021-11-19:
Fixed some problems with "no perimeter on bridge".
I may overhaul the feature for the v58, to be more useful.

closed 2021-12-22 by supermerill