supermerill/SuperSlicer #1921: One-loop perimeters not working on brim/raft/support

stuwe911, 2021-11-19:
**Describe the bug**
The main reason for me to change over to SuperSlicer was that magic "One-loop perimeters" setting which makes a massive difference when printing LW-PLA where any retraction is nearly impossible. This allows me to print parts in close to vase mode that were not designed for that. I have to use a lot of brims and rafts in order to get clean prints with LW-PLA. Over the last few weeks i found that this magic "One-loop perimeter" that truly works well in the part itself seems to be completely ignored when printing the brim/raft. resulting in the nozzle going from one end where the brim ended all the way across to the other end just to start the raft. totally messing up the first layer. seem was mostly set to "nearest" in the hope he would just start the raft where the brim ended like he does on the part itself. but this is not the case. the same weird back and forth goes on when printing supports.

**Expected behavior**
I'm not sure if this is a bug or per design. but it would make sense (to me) if the magic of One-loop perimeters would be included in the whole print from start to the end. regardless if you use brim/rafts/support especially on the first layer! A true game-changer to anybody printing LW-PLA

**Desktop:**
 - OS: Windows 7
 - Version 2.3.57.5

supermerill, 2021-11-19:
It's only for perimeter right now.
Do you have a project to share, so I can better understand the problem?

stuwe911, 2021-11-19:
shame.. but the magic is super cool and it would be fabulous if integrated in the whole print.
I've tried to attach a gcode but i get "we dont support that file"... how do i send you the code?
I've also noticed that on the raft when going 90degrees cris cross he only does that until layer 3. from then on it keeps printing onto the same line as layer 3. should i create a separate issue with more details for that?
thanks

supermerill, 2021-11-19:
in superslicer, "file" menu -> save project file
then zip it and drop it here. It contains the object & the settings to slice it.

stuwe911, 2021-11-19:
ok this might not be the best example but it should give you a idea what i'm talking about. if it's not good enough let me know and i can prepare another one.
cheers 

[F-18 Wing Tips.zip](https://github.com/supermerill/SuperSlicer/files/7573200/F-18.Wing.Tips.zip)