supermerill/SuperSlicer #1922: Fan speed setting for supported layers for easier removal of support material

pasyn, 2021-11-19:
Filaments such as nylon which have exceptional layer adhesion make it difficult to use mechanically removable supports. By bumping up the fan speed for the first layer of a part above a support can reduce the adhesion to the support thus making it easier to remove. Currently I am able to do this by changing the threshold for overhangs to use bridge fan speed and enabling the support bridges setting (see the attached screenshots). Ideally this could be implemented better as a separate setting.

[image]
[image]

supermerill, 2021-11-19:
set your bridge fan speed to 100%.
I don't understand really what's missing here.

pasyn, 2021-11-20:
Let me try to rephrase. What I am looking for is an option to adjust the fan speed specifically for paths that go over support material. I would like to adjust this separately from overhang and bridge fan speed. In my experience nylon can be finicky and too much fan can cause the part to warp therefore having this extra option will allow me to fine tune the adhesion between support and part to a point where the support can be removed and the part does not warp.

supermerill, 2021-11-21:
That may a bit difficult to add, as bridges are created before supports. And nothing links them.