supermerill/SuperSlicer #1925: Flow seems way too high or lines way too close in dense layer support

RomRider, 2021-11-19:
**Describe the bug**
With the last version 2.3.57.6 (and at least with the previous one too as far as I remember), the flow used for the feature "supporting dense layer" seems way too high. It seems to be at least 50% above what it should be.

See below for the project file.

Flow is orange, while it should be yellow (as the same speed is used for the other yellow lines and there is no gap between the lines)
[image]
As you can see, speed is the same.
[image]

It get even way worse for when printing the layer 8 (all the other layers above will blob)
[image]

Deactivating the supporting dense layer feature, give the expected results:
* with the setting activated:
  [image]

* without
  [image]

**To Reproduce**
Steps to reproduce the behavior:
See above.

**>> Project File <<**
[Calibration_OctogonV3.zip](https://github.com/supermerill/SuperSlicer/files/7571874/Calibration_OctogonV3.zip)

**Expected behavior**
The proper flow to be used with supporting dense layer feature active. The lines used to be separated which would make this flow rate logical, but not with the way it is now where all the line have no gap between them anymore.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 2.3.57.6

**Additional context**
Add any other context about the problem here.

supermerill, 2021-11-20:
should be fixed in next nightly

RomRider, 2021-11-20:
Amazing 😊 thanks!

closed 2021-12-22 by supermerill