supermerill/SuperSlicer #1926: Hi, do you have a profile for an Anycubic Predator?

carmichaeldap, 2021-11-19:
### Version
_Version of SuperSlicer used goes here (help->about)_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
* _Expected Results_
* _Actual Results_
  * _Screenshots if useful_

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_

supermerill, 2021-12-01:
Sorry, i don't have one.
You can try to search in an anycubic community.
You cana lso grab a prusaslicer one and use import-> from prusa.
If you have a good one, you can submit it here for inclusion.

carmichaeldap, 2021-12-01:
Thanks very much for the information. 
Cheers 
Dave

Sent from the all-new AOL app for Android