supermerill/SuperSlicer #1927: SuperSlicer crash if I use support and brim building

Jaz939, 2021-11-19:
Hi guys,

The SuperSlicer 2.3.57.5 crash (the application closes) if I tick the brim option and do support at the same time.
I'm attaching the slicer file to try it your self.

BR
Roman
[Andreja2.zip](https://github.com/supermerill/SuperSlicer/files/7572701/Andreja2.zip)
f.

supermerill, 2021-11-20:
If i load your project, it has the overlapping perimeter selected.
It slices without problem, as expected.
If I unselect it, it still slices...

macchinette, 2021-11-21:
If I can give my contribution, I tried to open the project and to finish slicing without crashing it is necessary to put the check mark in the parameter.

[image]

supermerill, 2021-11-21:
brim still have some issues with the support. I'm working on a fix for it.

supermerill, 2021-11-21:
not good enough, but at least it won't crash and should print.
Will improve it further

closed 2021-12-22 by supermerill