supermerill/SuperSlicer #1929: FR: Switch to Disable Automatic Processing

brotherdust, 2021-11-19:
**Is your feature request related to a problem? Please describe.**
Every time I adjust a parameter, SuperSlicer automatically starts re-slicing. This causes a slowdown in responsiveness in my system and is wasteful of compute resources (and energy). Background processing is nice, but it's still automatically triggered.

**Describe the solution you'd like**
A way to disable automatic processing and a way to trigger it manually _after_ I'm done tweaking everything.

**Describe how it would work**
When automatic processing is disabled, I have to manually trigger a re-slice. This will allow me to make batch changes to the parameters of my job without wasting a bunch of compute resources constantly resetting the slicing process.

**Describe alternatives you've considered**
Background processing; but it's still automatic.

**Additional context**
Automatic slicing is nice, but on large, complex models, it takes a LONG time to slice. I'd like more control of this process so I'm saving energy (battery on my laptop) as well as keeping my system responsive.

supermerill, 2021-11-19:
configuration (menu) -> preferences->General -> automation -> uncheck "background processing"

brotherdust, 2021-11-19:
I guess I was wrong. Thank you.

closed 2021-11-19 by brotherdust

supermerill, 2021-11-19:
I mean, the "slice button is a way to trigger the slicing when you are done with tweaking things.
I don't see how you can be between the two.

brotherdust, 2021-11-19:
@supermerill , don't sweat it. It's my bad. I'm sorry I wasted your time. =)