supermerill/SuperSlicer #1931: Missing include directives

Zombie-Feynman, 2021-11-20:
**Describe the bug**
I compiled SuperSlicer 2.3.57.6 on my Gentoo box (gcc-11.2.0, GNU gold linker) by adapting the PrusaSlicer ebuild, but I encountered a lot of compile errors due to missing includes. The following must be added before SuperSlicer will compile on this system:
`#include <boost/filesystem/path.hpp>` in `src/libslic3r/Format/CWS.cpp`
`#include <boost/filesystem/path.hpp>` in `src/slic3r/GUI/CalibrationAbstractDialog.cpp`
`#include <boost/filesystem/path.hpp>` in `src/slic3r/GUI/Preferences.cpp`
`#include <boost/filesystem.hpp>` in `src/libslic3r/Format/CWS.cpp`
`#include <boost/filesystem.hpp>` in `src/slic3r/GUI/CalibrationAbstractDialog.cpp`
`#include <boost/filesystem.hpp>` in `src/slic3r/GUI/Preferences.cpp`
`#include <boost/log/trivial.hpp>` in `src/libslic3r/Format/CWS.cpp`
`#include <boost/log/trivial.hpp>` in `src/libslic3r/GCode/AvoidCrossingPerimeters.cpp`
`#include <boost/log/trivial.hpp>` in `src/libslic3r/MedialAxis.cpp`
`#include <boost/log/trivial.hpp>` in `src/libslic3r/PlaceholderParser.cpp`
`#include <boost/log/trivial.hpp>` in `src/slic3r/GUI/PhysicalPrinterDialog.cpp`
`#include <boost/log/trivial.hpp>` in `src/slic3r/GUI/GUI_ObjectList.cpp`
`#include <boost/lexical_cast.hpp>` in `src/libslic3r/GCodeWriter.cpp`
`#include <boost/locale.hpp>` in `src/libslic3r/AppConfig.cpp`
`#include <regex>` in `src/libslic3r/PrintBase.cpp`
`#include <regex>` in `src/libslic3r/PlaceholderParser.cpp`
`#include <regex>` in `src/slic3r/GUI/Search.cpp`
`#include <unordered_set>` in `src/libslic3r/GCode.cpp`
`#include <atomic>` in `src/libslic3r/PrintObject.cpp`
`#include <boost/nowide/fstream.hpp>` in `src/slic3r/GUI/3DBed.cpp`
`#include <boost/property_tree/ptree.hpp>` in `src/slic3r/GUI/3DBed.cpp`
`#include <boost/property_tree/ptree.hpp>` in `src/slic3r/Utils/PresetUpdater.cpp`
`#include <boost/property_tree/ini_parser.hpp>` in `src/slic3r/GUI/3DBed.cpp`
`#include <boost/property_tree/json_parser.hpp>` in `src/slic3r/Utils/PresetUpdater.cpp`
`#include <boost/locale/generator.hpp>` in `src/slic3r/GUI/3DBed.cpp`
`#include <boost/lexical_cast.hpp>` in `src/slic3r/GUI/Tab.cpp`
`#include <wx/collpane.h>` in `src/slic3r/GUI/Tab.cpp`

**To Reproduce**
Steps to reproduce the behavior:
1. Download the source archive from https://github.com/supermerill/SuperSlicer/archive/refs/tags/2.3.57.6.tar.gz
2. Attempt to compile it on a Gentoo system (see attachment for toolchain versions, etcetera).

**>> Project File <<**
N/A

**Expected behavior**
SuperSlicer compiles successfully.

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: Gentoo Linux
 - Version 2.3.57.6

**Additional context**
See attachment for `emerge --info` output.

[emerge-info.log](https://github.com/supermerill/SuperSlicer/files/7574398/emerge-info.log)

supermerill, 2021-11-21:
will include

Tapchicoma, 2021-12-14:
I've added [an ebuild](https://gitweb.gentoo.org/repo/proj/guru.git/commit/?h=dev&id=bf759067f6e82e112b2d6f6d6ed0170fd7ed0679) for 2.3.57.7 version into [Gentoo Guru](https://wiki.gentoo.org/wiki/Project:GURU) repo. You could grab a patch for missing imports from there and just apply it.

Tapchicoma, 2022-01-26:
I've just added a new ebuild for version 2.3.57.9 and there is still one [missing import](https://gitweb.gentoo.org/repo/proj/guru.git/diff/media-gfx/superslicer/files/missing-imports-2.3.57.9.patch?h=dev) in `PrintObject.cpp`.

Also forgot to previously mention missing import for miniz_zip header that I need to apply as [a patch](https://gitweb.gentoo.org/repo/proj/guru.git/tree/media-gfx/superslicer/files/miniz-zip-header-2.2.53.4.patch).

Thank you for adding missing imports - adding an ebuild for the latest version was much easier then before :+1:

Tapchicoma, 2022-02-18:
Just added ebuild for 2.3.57.11 version and there is [one new missing import](https://gitweb.gentoo.org/repo/proj/guru.git/diff/media-gfx/superslicer/files/missing-imports-2.3.57.11.patch?h=dev).

Generally I would expect this issue to be ongoing with every new release. Probably I could just create a PR with fixes after each new release.

supermerill, 2022-02-18:
Yeah, compilers have different expectations.

> Probably I could just create a PR with fixes after each new release.

You can. I'll add this one for now.