supermerill/SuperSlicer #1932: Permanent ironing

rollduck, 2021-11-20:
**Describe the bug**
Hi,
I've changed Prusa slicer to a superslicer, and I have a problem, because the superslicer ironing the top layers even though this option is turned off. There is any other button to check out? 

**To Reproduce**

Just slice

**Project File**
NA

**Expected behavior**
possibility to turn on/off ironing

**Screenshots**

[image]
[image]

**Desktop (please complete the following information):**
 - OS: Win10
 - Version 2.3.57.6

**Additional context**
NA

rollduck, 2021-11-20:
Ah ok, I changed the parameter "ironing infill pattern tuning" to 0 and it works, but it is strange for me, I thought that "enable ironing post-process" is a master option.

supermerill, 2021-11-20:
you just had the top pattern set to "ironing".
In SuSi, you have two way to do ironing, the ols one use a pattern (can also be used on support interface, useful for soluble one)
and there is the new prusa way, renamed 'ironing post-process'.

d-schmidt, 2021-11-20:
> Ah ok, I changed the parameter "ironing infill pattern tuning" to 0 and it works, but it is strange for me, I thought that "enable ironing post-process" is a master option.

Just change the ironing infill to monotonic filled.
[image]

rollduck, 2021-11-20:
You're right I didn't notice this option, many thanks for your help.

bagge, 2022-05-10:
I am still having issues with this on
OS: MacOS 11.5.2
Version 2.3.57.12

Infill->Top = Monotonic
Infill->Enable ironing post-process = False

When I slice I still get ironing done. The only way I found to get rid of the ironing is to:
1. Infill->Top = Ironing (To enable fill_smooth_distribution)
2. Infill->Ironing infill pattern tuning, Distribution = 0
3. Infill-> Top = Monotonic

Now if I slice I don't get ironing as expected. Though it seems to me that it is a bug that fill_smooth_distribution gets used even when ironing is not selected for top surfaces, wouldn't you agree?

bagge, 2022-05-10:
Actually, if I do the above I get rid of ironing in the features view of the platter. But the print time is unchanged and ironing is still in the exported gcode. Can't seem to find any way to actually suppress ironing...

SY573M404, 2022-11-23:
@bagge are you using generated calibration parts? I've ran into the same issue, and the cause were per model settings, which set top fill pattern to Ironing