supermerill/SuperSlicer #1934: Crash after slicing done (lithophane)

tome9111991, 2021-11-20:
Hello
i try to slice a lithophane it is a big stl file.
but the slicer will crash before i can export the G-Code.
on bottom left from susi show me the messsage slicing but bit later crash without error.
i can't the information about printing time etc... 

but only happens if unchecked the option "Overlapping perimeter" "Also for all perimeter"
[image]

if i check "Also for all perimeter" i dont have a crash.

Maybe you can find the problem.

Here project file but is to big for github:
https://file.mail91.de/f/4a84933af03c4cd2871e/?dl=1

supermerill, 2021-11-20:
If i load your project, it has the overlapping perimeter selected.
It slices without problem, as expected.
If I unselect it, it still slices...

tome9111991, 2021-11-21:
Hm okay maybe is my PC to bad.
I will try in linux if i have time

tome9111991, 2021-11-21:
> If i load your project, it has the overlapping perimeter selected. It slices without problem, as expected. If I unselect it, it still slices...

Btw.
I tought the option
"Also for all perimeter" it is unselect in my project.
but it still seems to work for you

supermerill, 2021-11-21:
maybe a ram issue? how many have you?

tome9111991, 2021-11-21:
have 12gb
[image]

supermerill, 2021-11-21:
11go used on my windows.

tome9111991, 2021-11-21:
maybe i can test on a other PC but only in a few weeks.

if you want you can test with my configdir:
[data.zip](https://github.com/supermerill/SuperSlicer/files/7576860/data.zip)

supermerill, 2021-11-21:
I mean, your computer only has 12go and the slicer need 11go, so I think it's why it crash: not enough memory.
It "only" needs 8go if you slice it on the command prompt, as it doesn't need to create the 3D view.
You can also reduce the precision of the slicing (in the print->slice tab) to prevent too much details.

tome9111991, 2021-11-21:
Ok thanks u will try.

Is there a guide how to use the command prompt?

nandobarreto, 2021-11-21:
> I mean, your computer only has 12go and the slicer need 11go, so I think it's why it crash: not enough memory. It "only" needs 8go if you slice it on the command prompt, as it doesn't need to create [...]

I dont think this is the case, i'm trying to slice this: https://www.thingiverse.com/thing:4102658 (10 inch version).
I have 64 GB of ram and i'm having the same issue, chashs after 2 or 3 minutes in the slicing.

tome9111991, 2021-11-21:
i have sliced the .3mf with superslicer_console.

superslicer_console will use the settings/profil what saved in the .3mf project file?

supermerill, 2021-11-21:
> Is there a guide how to use the command prompt?

https://manual.slic3r.org/advanced/command-line

> superslicer_console will use the settings/profil saved in the .3mf project file?

Yes, If you load a .3mf, it should load the config inside it unless it can't find it inside.

@nandobarreto 
can you provide me your project file (saved just before you click on "slice") , so I can have the exact same configuration?

nandobarreto, 2021-11-21:
> https://manual.slic3r.org/advanced/command-line
>
>
> Yes, If you load a .3mf, it should load the config inside it unless it can't find it inside.
>
> @nandobarreto can you provide me your project file [...]

https://www.dropbox.com/s/1ccdgpyvfqzo6fn/moon_lamp_10_inches_refinement_level_360.3mf?dl=0

please let me know after you get so I can delete.

supermerill, 2021-11-23:
@nandobarreto downloaded, Didn't had the time to test yet.

closed 2022-03-22 by tome9111991