supermerill/SuperSlicer #1935: Slicing is totally broken.

Miwer, 2021-11-20:
### Version
Version 2.3.57.6

### Operating system type + version
Operating System: Manjaro Linux
KDE Plasma Version: 5.22.5
KDE Frameworks Version: 5.87.0
Qt Version: 5.15.2
Kernel Version: 5.14.10-1-MANJARO (64-bit)
Graphics Platform: X11
Graphics Processor: Mesa DRI Intel® HD Graphics 4000

### Behavior
Slicing is totally fubar'ed. I have no idea what is wrong here, maybe someone have an idea, or it is a bug in the program?
After loading a model (or in this case, a calibration item), **preview shows correctly sliced for a split second**, but then changes to a completely useless slicing. Printing is also wrong, loading the gcode in another program shows the garbled version.

Screenshots below, but please see screenrecording (mkv) for the real effect:
[image]
[image]

#### Project File (.3MF) where problem occurs
Attaching 3MF file, gcode file, and a screenrecording showing the issue. All zipped

[Bridge calibration.3mf.zip](https://github.com/supermerill/SuperSlicer/files/7575083/Bridge.calibration.3mf.zip)
[Lulzbot Mini_Bridge calibration_0.3_ABS_2h14m.gcode.zip](https://github.com/supermerill/SuperSlicer/files/7575084/Lulzbot.Mini_Bridge.calibration_0.3_ABS_2h14m.gcode.zip)
[simplescreenrecorder-2021-11-20_20.02.43.mkv.zip](https://github.com/supermerill/SuperSlicer/files/7575085/simplescreenrecorder-2021-11-20_20.02.43.mkv.zip)

Best regards,
/Miwer

supermerill, 2021-11-20:
Don't set the gcode precision to 0, it means that you want no decimals.
it's intended to qhow bad things i this case, but it's many too bad to be allowed.

Miwer, 2021-11-20:
Oh my god, was it only that?  It worked.
I think I must have changed that value from the default by accident (mouse slip, or something), since I have no memory of wanting to change that.

I'm closing this again. Thanks :)