supermerill/SuperSlicer #1936: Problem bridging when using a sacrificial layer of the same height.

BoothyBoothy, 2021-11-20:
### Version
SuperSlicer Version 2.3.57

### Operating system type + version
Window 10 64 bit

I am relatively new to super slicer but cannot find anything online regards this so thought It may be worth mentioning it. Apologies if this is just newbie ignorance. This is an awesome slicer!

I use a lot of bridging and sacrificial layers when designing parts to reduce the need for supports.

When slicing with Super Slicer the Bridging retains the adjacent perimeters if a sacrificial layer is set to the same height as the rest of the part making the sacrificial layer useless. If I reduce the sacrificial layer by only 0.1 mm the bridging works fine.

It is logical to design the part with all adjacent layer heights to be the same height. It just complicates things if I have to set my designs to one specific slicer limits when publishing designs.

Is there a work around to this in the settings that I have not noticed?

Apologies again if this is not an issue and just down to my ignorance.

Best wishes Boothy

Bridging with same height sacrificial layers to the holes (0.25 mm layer height)
[image]

Bridging with sacrificial layers set to the holes 0.1 mm less than surrounding part @ 0.24 mm layer height.
[image]

supermerill, 2021-11-20:
duplicate of #1917

also of #1911 #1867

if you want auto-build sacrificial layer, use the nightly.
if you want man-made one, disable "only one peri top" for now, at least on these layers.

BoothyBoothy, 2021-11-20:
Many thanks for a fast response and apologies for my duplication. I will try  your suggestions. Boothy

BoothyBoothy, 2021-11-20:
This work very well, thank you again. Boothy

closed 2021-11-20 by BoothyBoothy