supermerill/SuperSlicer #1937: Dropdownbox expands out of view.

fmjensen, 2021-11-21:
### Version
2.3.57.5

### Operating system type + version
Windows 10 Pro, 20H2

### Behavior
- The Dropdownbox for "Show Options" in bottom left corner of the Gcode preview tab expands down and outside screen on secondary monitor.
- This does not happen if SuperSlicer is moved to primary monitor - here the Dropdownbox expand upwards.
- The dropdownbox for selecting what to "View" is not affected of this minor bug and expands upwards as it should on both primary and secondary monitor.

_Is this a new feature request?_
No.

#### Project File (.3MF) where problem occurs
- Not related to project files so not file are uploaded.

supermerill, 2021-11-21:
erf. Your second monitor is higher up than the primary?

I don't know if I have any control over that (there's nothing obvious).
I don't want to begin hacking the GUI lib...

neophyl, 2021-11-21:
There was https://github.com/prusa3d/PrusaSlicer/issues/2999 which was supposed to have been fixed in 2.3.1rc by fixing wxwidgets.  Could it have popped back up ? or the merge might have broken it ?

supermerill, 2021-11-21:
mmm. They forked wxwidget (the gui library).
I guess I have to modify the building script to use their fork instead of the main one...