supermerill/SuperSlicer #1938: Internal error: ConfigBase::get_abs_value(): nozzle_diameter has not a valid option type for get_abs_value()

gitcnd, 2021-11-21:
**Describe the bug**
Pops up the error: Internal error: ConfigBase::get_abs_value(): nozzle_diameter has not a valid option type for get_abs_value()

**To Reproduce**
Steps to reproduce the behavior:
1. New install
2. Define new printer with platform png, platform stl, and 0.6mm nozzle.
3. error pops up right after

**Expected behavior**
Understands 0.6mm nozzles!

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - MacOS
 - High Sierra

**Additional context**
First time user.
Also - annoyed that AnyCubic has lots of printers, but not the Chiron listed!

supermerill, 2021-11-21:
Can't reproduce.
My steps:
downloaded a 2.3.57.6
create a configuration directory to be sure it's empty
launch.
closed the wizard
saved the printer profile to untitled
add a random bed texture & stl found in the resources/profile
changed the nozzle from 0.4 to 0.6
no crashs yet, so I added a cube and sliced it. Still no crash.

gitcnd, 2021-11-21:
Might be a better idea to look at the code first?  It's telling you exactly what broke.

Assuming some settings got saved before the crash - let me know where those might have been so I can send over the files.

gitcnd, 2021-11-21:
p.s. - you should not have done the step "create a configuration directory to be sure it's empty" - that's cheating.
"closed the wizard" is beyond the bug already - the code crashes (exits) *in* the wizard.
FYI1 - I've never installed/run this before.
FYI2 - I placed the program into my Applications folder, and "allowed" it to run from the security settings (maybe you should find someone to sign your binaries?  That step probably loses more than half your mac users... - all the noobs who don't know how to run unsigned code, and all the security-conscious people who never run unsigned code).   unsigned stuff is a magnet for malware/ransomware/evil to attach, if you weren't aware....

supermerill, 2021-11-21:
Oh, you mean using the "custom printer" wizard.
I retried with it.
- opening superslicer with empty configuration
- in the wizard, select "custom printer"
- checking the box, entering a random name
- next 
- next 
- selecting a texture and a stl from the resources.
- next
- change nozzle to 0.6
- finish
still no problems.

kernelPanic0x, 2021-11-28:
I have the same problem as described by Chris. The error only gets triggered when I switch to another tab from the "Printer Settings" tab. In the printer settings i can change everything without a problem like custom start gcode etc. But when I switch tabs than the error message pops up.

supermerill, 2021-12-01:
@kernelPanic0x

Still can't reproduce.
Can you explain each step you did that lead to the issue?

gitcnd, 2021-12-01:
You are not supposed to reproduce bugs - you're supposed to find the problem, desing your own test for that, check the test triggers the bug, fix the bug, and run your test to check it's fixed.

ONLY AFTER THE ABOVE, do you return the fix to the user for THEM to check that you fixed it.

You are NEVER supposed to use the actual bug report, as the test.

This is basic QA-101 science, put in place to prevent coders from fixing symptoms instead of problems...

Not to mention common sense - you've wasted more of our time and yours on this easy-to-find-and-fix issue, than it would have taken to look at the code and spot the mistake in the first place... the error-message does tell you *exactly* what broke you know...

DO NOT FLAME ME: I'm and expert maintenance engineer, educating you FOR FREE on the correct way to do things.  Pay attention and be thankful that I invested my own time in helping you learn important lessons.

neophyl, 2021-12-01:
I respectfully disagree.
When producing a defect report a test engineer (or in this case a volunteer supplying their time and effort) is supposed to describe the steps needed for the developer/software team to reproduce the issue.  Once the team can reproduce the issue then they can confirm the issue is real.  In this case the team is a team of one.  They can then modify their code and engineering test that the change or changes eliminates the issue.  
They do NOT go blindly look around the code unless that code is trivially simple.  Slicer software is patently not simple.

Normally they then pass on the code for the test team to confirm that the defect has been fixed.  The testing methodology depends on the software and should not exactly duplicate the engineering tests done.  Once again as its a team of one then I don't think there's much if any separation between engineering/formal testing.  Its pretty much the user base doing final testing.  

At the moment the developer hasn't even managed to reproduce the issue so can't confirm its existence.  To me that means the described steps are either not precise enough or there is some missing factor between the systems that is causing the issue.  Until that factor can be identified then its impossible to know if any software change would result in a fix.  Cant test if something is fixed if you can't duplicate the problem.

Oh and btw I am a senior software test engineer that has been testing embedded systems for the last 20+ years.  

That's not a flame btw, that's my opinion on the issue  from the information presented here.

One thing you you ought to remember though is the Merill is ALSO doing this for FREE and that your 'correct' way is not the only way of doing things, just the way that you may have been taught to do them.  Your last sentence comes over as incredibly condescending and arrogant.  Even if you don't mean it that way.

gitcnd, 2021-12-02:
It looks like you don't have a software degree, and you definitely don't understand maintenance or the concept of "Quality Assurance".

Everything you said is totally wrong.

When a customer reports an issue, it's probably real (**especially** when it includes a screenshot of the software actually saying it is) - when 2 different customers independently report the same issue, the "realness" is 100% confirmed.  Regardless of either of those - it's up to the developer to investigate/fix the issue, **not** to try and prove a customer is wrong.

I added the "flame" comment because "feelings" do not belong in maintenance.  My 37 years beasts your 20 years, not that it's relevant since I learned the "right way" before you even started doing it the wrong way in the first place.

Also remember - that as soon as you attack the messenger (e.g. "incredibly condescending and arrogant"), instead of staying on-topic, that's the psychological indicator that you already understand you're wrong, and you're now looking for some other way to win the argument.

DO NOT REPLY.  This is not an argument, and you "agreement" is irrelevant.  There's a line in your code defining "get_abs_value" which wants your attention.

kernelPanic0x, 2021-12-04:
> @kernelPanic0x
> 
> Still can't reproduce. Can you explain each step you did that lead to the issue?
> 
> or a video.

What I've done to encounter the issue:

-> Removed all configurations created from SuperSlicer (removed %appdata%/superslicer folde)
-> Fresh copy unzipped and started for the first time
-> Assistent pops up
-> Choosing custom printer
-> Giving it a name, Settings Marlin, Settings bed size
-> ! Settings nozzle diameter to 0.60 mm !
-> Settings bed temperature to 60°C (rest default)
-> MAYBE IMPORTANT: Tagged Export Visibiolity Mode! Rest stays unchecked.
-> Quit Assistent
-> Not changing to Printer Settings tag.
-> As soon as I change from the Printer Settings Tab to the Print Settings Tab (third from right) I get the "Unhandled unknown exception; terminating the application" error.

Thats it. Tried it several times with different scenarios when I put 0.60 nozzle diameter and change the tab combination it triggers the error message.

pfakanator, 2021-12-11:
I encounter the same issue only if I enable all filament types during the wizard.

supermerill, 2021-12-11:
@pfakanator 
 even on the nightly?

pfakanator, 2021-12-11:
Saw the commit but haven’t used nightly yet.  Just relaying my repeatable experience.

pfakanator, 2021-12-12:
I cannot reproduce this with the latest release.  Looks squashed.

supermerill, 2021-12-13:
good

closed 2021-12-13 by supermerill