supermerill/SuperSlicer #1939: Bridges infill not following overhangs perimeter direction

SellA, 2021-11-21:
**Describe the bug**
Slicing a particular file that has bridges in 2 directions, all are sliced in the same direction, failing following the perimeters path.

**To Reproduce**
Steps to reproduce the behavior:
1. Open the attached .3mf file
2. Slice it
3. Preview around layer height 10.8mm and look for the first layer of bridges
4. Look at the bridges on the right: overhang perimeters are correctly layed down, bridge infill is turned 90° in the wrong direction.
5. Also bridge overlap is set to 100% but seems that there is some spacing between the extrusions. This is noticeable on the preview, maybe they are correct but that is tricking me.

**>> Project File <<**
[Bridges bug report.zip](https://github.com/supermerill/SuperSlicer/files/7576667/Bridges.bug.report.zip)

**Expected behavior**
Bridges infill should be layed down in the same direction as  overhang perimeters because extrusion must have contact with the solid layers layed on earlier layer. Overhangs perimeters are just "overhangs" that are not squished into previous layers so their width is not large as other extrusion and can't hold infill correctly.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 21H1
 - Version: 2.3.57.

SellA, 2021-11-21:
So, i found that "Briding angle" was set to 90° and it was forcing the direction on the bridging. Setting it to 0 solved my issue. What is still strange is the bridge infill. Increasing the "bridged_infill_margin" parameters from 200% to 300% i get more lines of bridging,  closing completely all the gaps, also overlapping some of the overhang perimeter. I attach a new screenshot. I don't understand why this parameter influece the number of lines layed down for bridging.
[image]

supermerill, 2021-11-21:
> I don't understand why this parameter influece the number of lines layed down for bridging.

It can change the bounding box used to compute the bridge position & spacing.

Note that the bridge position is updated in the current nightly, so the current behaviour may be updated, hopefully in a good way.

SellA, 2021-11-21:
Ok so in nightly build i see the correct behaviour. With 200% "bridged_infill_margin" the overhangs are completely filled also there is some overlap that i think is good. I'll try to print it tomorrow and report back and hopefully close the issue!

[image]

SellA, 2021-11-22:
[image]

Bridges printed perfectly on latest nightly!
Thanks for the support, glad that super slicer keeps improving!

closed 2021-11-22 by SellA