supermerill/SuperSlicer #1940: [FR] Option to enable both M117 and M73 progress reporting at the same time

kb-elmo, 2021-11-21:
**Is your feature request related to a problem? Please describe.**
No issue. Would just be a nice to have thing

**Describe the solution you'd like**
Add the option to enable both M117 and M73 reporting at the same time.

[image]

**Describe how it would work**
Send the M117 display message and M73 percentage report command together in the Gcode.

supermerill, 2021-11-21:
It's useful? i had no idea.

kb-elmo, 2022-01-18:
will this get added? 😄 

> It's useful? i had no idea.

it would be for me (and a few other people that I know)

supermerill, 2022-03-11:
I would have liked an explanation of the use-case.

kb-elmo, 2022-03-16:
I thought it was pretty obvious 😄 

The standard display for Marlin shows both a text message and the progress bar.
You need both options to make use of those.

Otherwise either the message text or the progress bar will stay empty.

[image]

closed 2022-07-14 by supermerill