supermerill/SuperSlicer #1942: We need more Purge!!

Durahl, 2021-11-21:
Greetings! 😃

I'm currently trying to print a Multicolor part, but the two colors appear to be at odds with each other - Black still keeps bleeding into red despite me already having red's purge volume raised from 200 to 300 - With me now attempting to raise it even further to 400 I've now noticed the values being capped at 300? 😥

Any suggestions other than duplicating the part and using one as a sacrifice for the other to get a little more material purged?

DzzD, 2021-11-22:
Sorry I have no direct answer but here are some informations & tips :

First I would say that black or white are the worst case in multicolor so if you succeed with one of this color you will be ready for any color.

You can try to add an ooze shield with a thickness of two or more outlines that will purge even more color.
[image]

If you can and and it is the best solution for multicolor with one share nozle, add a purgebucket to your printer, it is possible on some printer not all, google for it.

Durahl, 2021-11-22:
Mhh... Gonna have to give this a thought.

The developer of the [ERCF](https://github.com/EtteGit/EnragedRabbitProject) I'm using for Multifilament prints is still working on a purge bucket solution that has yet to be released.
Out of curiosity though, what is the benefit of a Purge Bucket System over a Wipe Tower to begin with?
Is it not bound to such settings limitations? 🤔

Durahl, 2021-11-30:
I've since found out that one can get higher purge volumes out of a given print via the use of the _"Advanced wipe tower purge volume calculs"_ in the **Printer settings** and setting the necessary pigmentation values for the respective filament 🤔

supermerill, 2021-12-01:
Did you try to click on "advanced settings" ?

[image]

Durahl, 2021-12-02:
I did not - For one I assumed it having the same limitation and for the other I'm having a hard time understanding how the array even works to begin with 😅

supermerill, 2021-12-02:
> I'm having a hard time understanding how the array even works

your current color is : choose your line
and you are changing for the color : choose the column
then, at the intersection, set the desired amount to purge.

Durahl, 2021-12-03:
[image]
Would this make sense? 🤔
I assume it does - Wipe tower appears to be purging a lot of red and orange with little black but just making sure 😉

supermerill, 2021-12-03:
if you go from Black to Red, you extrude 400mm (line1/2 -> column 3)
if you go from yellow to black, you extrude 30mm. (line 5, column 1 or 2)
if you go from grey to orange, you extrude 150mm (line 6, column 4)

closed 2021-12-14 by Durahl