supermerill/SuperSlicer #1945: Enabling supporting dense layer is not 50% infill after 2.3.57.3

samwiseg0, 2021-11-22:
**Describe the bug**
I recently updated to 2.3.57.6 from 2.3.57.3 and I noticed that after slicing the bridging layer looked very different from 2.3.57.3. I also tested 2.3.57.4 and it seems the bug may have been introduced in that version. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Printer Settings
2. Enable supporting dense layer
3. Slice

**>> Project File <<**
[mag_board_marker_shelf_v2.3mf.zip](https://github.com/supermerill/SuperSlicer/files/7579307/mag_board_marker_shelf_v2.3mf.zip)

**Expected behavior**
I believe the expected behavior is what I experienced with 2.3.57.3.

**Screenshots**
**Setting:**
[image]

**2.3.57.3:**
[image]

**2.3.57.4 - 2.3.57.6:*
[image]

**Desktop (please complete the following information):**
 - OS: MacOS
 - Version: 2.3.57.4+

**Additional context**
NA

supermerill, 2021-11-23:
duplicate of #1925
fixed in the nightly

samwiseg0, 2021-11-23:
> duplicate of #1925
> fixed in the nightly

Great! Thank you

closed 2021-11-23 by samwiseg0