supermerill/SuperSlicer #1948: Print speed < 1mm/s for small_perimeter_speed while 60mm/s is set

Sverto, 2021-11-22:
**Describe the bug**
With several prints I've noticed a perimeter speed of less than 1mm/s resulting in a failed print.
The default perimeter speed is set to auto.
The "small_perimeter_speed" value is set to 60mm/s.
Thin walls is turned off (in case it's relevant).
The minimum filament layer cooling speed is set to 10mm/s.

I've tried to find the cause and it seems that 'when a perimeter falls into the small_perimeter_speed category' its speed is reduced to < 1mm/s while the actual "small_perimeter_speed" value is set to 60mm/s. 

**To Reproduce**
Steps to reproduce the behavior:
1. Set perimeter speed to auto
2. Slice an object with a perimeter that falls into the small perimeter category
3. Notice the inexplicable slow speeds for these perimeters

**>> Project File <<**
[superslicer_bug.zip](https://github.com/supermerill/SuperSlicer/files/7583129/superslicer_bug.zip)

**Expected behavior**
The configured small perimeter speed value to be respected (or minimum layer cooldown speed).

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version: 2.3.57.4

Could this be verified?
Thank you in advance.

supermerill, 2021-11-23:
There is something weird.

TokkenStrike, 2021-11-26:
Came here to see if anyone else had this issue.
I got the speed below 0.1mm/s and only noticed it when the printer actually stoped moving (at least I didn't see it move) in the middle of the print.
Also lowerging min length decrease the speed (was expecting it to increase but I may be wrong...), and matching min and max length use the right speed.

supermerill, 2021-12-03:
Found the problem. I applied the slowdown below the '0' was replaced.

closed 2021-12-22 by supermerill

afbenevides, 2022-02-10:
Hello, I know this one is closed, but I think I may be seeing the same/similar problem while on version 2.3.57.10_2022-02-03 (on OSX)

Maybe one of you can comment on that?

Here is what I get with different settings from 0, percentage % and mm/s setting. Maybe I did not understand well the auto functionality for this parameter and that it refers to another parameter that is wrong, but I hope someone can point me out where to look for that fix.

Would it be looking for the 0 mm/s (for auto parameter of perimeter_speed) for the speeds parameters in the top of same page instead of the resulting speed of the automatic calculated speed ?

Thanks in advance

0 (auto auto setting)
[image]
[image]

50% setting
[image]
[image]

50mm/s setting
[image]
[image]

supermerill, 2022-02-10:
@afbenevides can you post your project file?

afbenevides, 2022-02-10:
[example_ultra_low_speed_small_peremeter.3mf.zip](https://github.com/supermerill/SuperSlicer/files/8042747/example_ultra_low_speed_small_peremeter.3mf.zip)

Seafnox, 2023-03-09:
Still Have that issue on Version 2.5.59.2

With Default speed - 120 mm/s

When External perimeter speed is 50% - real speed in thin perimeter is 11.4
When External perimeter speed is 100% - real speed in thin perimeter is 12
When External perimeter speed is 500% - real speed in thin perimeter is 87
When External perimeter speed is 1000% - real speed in thin perimeter is 147

Small perimeter speed parametes is no weight on it (10%/50%/100%/500% - isn't matter)

[image]
[image]

Seafnox, 2023-03-09:
Here my project
[Stacking_Box_-_0.5mm_wall_thickness.3mf.zip](https://github.com/supermerill/SuperSlicer/files/10933669/Stacking_Box_-_0.5mm_wall_thickness.3mf.zip)

Seafnox, 2023-03-12:
Found the reasons - too high layer time brings to low printing speed
[image]