supermerill/SuperSlicer #1949: FR: Seam Position - Overhang Cost

CCS86, 2021-11-23:
Positioning seams on areas which have a large overhang angle can be problematic. Having a cost factor for overhang could help the cost based algorithm avoid placing seams in overhung areas, when a more vertical wall might exist.

supermerill, 2021-11-23:
I thought they can't place a seam in an overhang.
Can you give me a project file where this happens?
(file -> save project as  then zip it and drop it here)

CCS86, 2021-11-26:
I must have been mistaken. Was watching a print and though I saw it making seams on an overhang. I haven't been able to reproduce in SS. If seams on overhangs are already disallowed globally, we can close this request.

closed 2021-11-26 by CCS86