supermerill/SuperSlicer #1953: Flow Calibration Wizard drops Temp to 0

BlueDun99, 2021-11-23:
Trying to use the flow calibration wizard. After my initial routines the temperature setpoint drops to 0.
Could not find an M104 or M109 in the code that could be responsible. However, found G1 command lines with [A] Parameter - never seen before (but I'm not a G-code pro). Lines 63 and 68 in the code.

**Additional context**
Add any other context about the problem here
[Flow calibration_0.2mm_PLA_ENDER3BLTOUCH_26m.gcode.zip](https://github.com/supermerill/SuperSlicer/files/7587043/Flow.calibration_0.2mm_PLA_ENDER3BLTOUCH_26m.gcode.zip)
.

supermerill, 2021-11-23:
your project file?
I need it to be able to see your config & how it interact with the objects.

BlueDun99, 2021-11-23:
Sorry .. the project file would indeed make sense ...
I did not save the project and now I'm trying to recreate it. But I cannot reproduce the bug. The job finishes without problem and the A parameter for the G1 commands are gone and replaced by E and Z.
I guess case can be closed ... Apologies for the trouble

closed 2021-12-01 by supermerill