supermerill/SuperSlicer #1954: Separate speed settings for first layer Brim

stuwe911, 2021-11-23:
**Is your feature request related to a problem? Please describe.**
I print large objects with tiny details on the first layer. I do require a large brim in order to avoid lifting or massively reduce the overall printing speed. The current Brim speed is set by the first layer speed so it does not differentiate between printing the brim or the first layer of the actual print. Printing large Brim's at low speed takes forever and then it's usually too fast for the details on the first layer of the print. Currently my only solution is to manually change the speeds accordingly.

**Describe the solution you'd like**
Have a separate speed setting for the Brim (optional Skirt/Raft as well) than the first layer of the print. I can print my Brim's at 100ms where the print only takes 10ms. Allowing the first layer's speed setting to be split into actual print and support material since the Brim is usually more simple but larger and can run at faster speed. This would allow to print larger Brims for a better overall success of the print.

**Describe how it would work**
Add a "Brim speed" field

**Describe alternatives you've considered**
Manual override of the speed whilst monitoring the first layer. guessing when to change and waiting for the printer to accept the new settings.

Additionally but not super urgent would be a "number of layers" of the Brim like many other slicers have in order to make the Brim stronger

Shampoo1014, 2022-04-21:
@supermerill Any update on this? I was about to make a feature request when I came across this one. I would love to have a separate speed for my skirts. 
Thank you for the amazing work on SS! It's changed my entire printing life.

Shampoo1014, 2022-04-25:
> I just added a brim& skirt speed & acceleration settings.
> #1789 <https://github.com/supermerill/SuperSlicer/issues/1789>

Thanks so much Merill, looking forward to playing around with it. Keep up
the good work.

wnc2, 2022-06-22:
i'm not able to vary brim speed, everything i write, integer or % it prints always at 30mm/s
i don't have limits
[image]

MoHub1, 2022-06-25:
Hi, I'm having the same issue where the brim and skirt speeds seem to stay at 30mm/s - changing the Brim & Skirt speed setting doesn't appear to have any affect

StuweFPV, 2022-09-18:
I found that changing the brim and print speed on the 1st layer does not have any effect. however if (in the rare occasion) that i make my brim 2 layers in hight or more THEN the new setting is grasping. So it's partially working. from layer 2 onwards it works fine. But I guess we all want it to work on layer 1. Still a big thank you @supermerill

CFK520, 2023-04-18:
same +1