supermerill/SuperSlicer #1955: [FR] Enhance slowdown_below_layer_time

pockata, 2021-11-23:
**Is your feature request related to a problem? Please describe.**
It's more of an enhancement suggestion than an issue. It's best observed in the [Cali Cat model](https://www.thingiverse.com/thing:1545913).

The slicer slows down print speed if the layer time is below the configured amount, however it does little or no account for the fact that part of the layer is an island (tower), that is separated from the main part of the object. In the examples below the tail of the cat is counted towards the total layer time and its printing speed is the same as the body of the cat which in Slice 1 is bad, because the whole print is slow. In Slice 2 the body of the cat is printed with a normal speed and turns out perfectly, however the tail of the cat is messy because each previous layer didn't have enough time to cool down from the part cooling fan before the next layer was printed.

Essentially the body prints perfectly at 30 and at 80mm/s, however the tails prints perfectly only at around 25-30mm/s.

### Slice 1

**Filament settings**:
[image]

**Print settings**:
[image]

**Sliced model**:
[image]

### Slice 2

**Filament settings**:
[image]

**Print settings**:
No change

**Sliced model**:
[image]

**Describe the solution you'd like**
A modifier/alternative for the `slowdown_below_layer_time` option that takes the object's print area in into account (in mm²).

**Describe how it would work**
Instead of using the layer time (which might not be ideal, as shown in the examples above), use the calculated area (in mm²) for each island and slow down print speeds only where necessary.

Using the Cali Cat example, the print speed should slow down only for the tail, and use a consistent speed for the cat's body and head (not like in the modifier mesh example below).

**Describe alternatives you've considered**
As a workaround I've used modifier meshes to slow down the print speed of the tail, however there are a few drawbacks to this approach:
1. The tail is still counted towards the total layer time and this affects the print speed of the body versus the print speed of the head (where there's no tail). It would be better to print the cat body's outer perimeters with one single speed.
2. This approach requires per-print modifications. It's not a tune and forget option.
3. Might have issues when printing multiple parts, but I rarely do that and can't really comment on it.

[image]

**Additional context**
I'm printing it on a Hypercube (corexy) running Klipper, using PLA and a 5015 part cooling fan at 100%

royal32, 2021-11-23:
I've thought about this before, I agree. The minimum layer time speed & cooling features should be based on the islands of each layer and not the entire layer. Would be a worthwhile change IMO.

macchinette, 2021-11-23:
I agree with what is indicated as an improvement, because the closed perimeters and therefore isolated from the other perimeters require different speeds and ventilation.
At the moment I found a trick with the parameters small_perimeter_min_length and small_perimeter_max_length

[image]

supermerill, 2021-11-23:
the slowdown is here to let the previous layer cool down. The problem here is that it stay on the small island when going to the next layer, instead to begin agin on the cat part, so the small tail is still too hot?

pockata, 2021-11-24:
> I agree with what is indicated as an improvement, because the closed perimeters and therefore isolated from the other perimeters require different speeds and ventilation. At the moment I found a trick with the parameters small_perimeter_min_length and small_perimeter_max_length

@macchinette, brilliant! The `Small perimeter speed` settings provide a great alternative to the modifier mesh approach and as far as I can see from my limited tests, when coupled with a disabled `slowdown_below_layer_time` setting they provide pretty much all of the functionality I'm suggesting.

> the slowdown is here to let the previous layer cool down. The problem here is that it stay on the small island when going to the next layer, instead to begin agin on the cat part, so the small tail is still too hot?

@supermerill Not exactly. The next layer's starting points alternate between the cat's body and it's tail. It's working as expected.

What I'm trying to illustrate is that `slowdown_below_layer_time` can be improved because it doesn't take small perimeters (islands) into account. It slows down parts of the print that don't need it (the cat's body).

I've achieved the desired effect with @macchinette's suggestion. Abusing `Small perimeter speed` yields the same results I wanted to achieve with my initial suggestion. The only difference is that the setting uses the perimeter length and not the calculated area like I suggested. 

By 'abuse' I mean you just have to set `small_perimeter_min_length` and `small_perimeter_max_length` to the same value so you can explicitly use `small_perimeter_speed`, otherwise an unspecified slowdown is applied to some perimeters.

Here are the new settings I'm using and the result they yield:

**Filament settings:**
[image]

**Print settings:**
[image]

**Sliced result:**
[image]

As you can see, with these settings (no modifier meshes!) the model is sliced way better than with any of my tests before. It slows down print speeds only where needed (the tail) and has a consistent speed across the whole body of the cat.

With these new findings I'm required to change my initial suggestion.

`slowdown_below_layer_time` and the `Small perimeter speed` settings ultimately try to achieve much of the same goals, but they both fall short. More accurately, `slowdown_below_layer_time` is flawed (it creates unnecessarily slow prints/uneven print speeds), and `small_perimeter_max_length`'s "Every perimeter loop lower than this will see their speed reduced a bit" seems like it's taking away control out of the hands of the user instead of giving us more control.

So as a solution I propose the following changes:
1. Remove the `Very short layer time` section altogether
[image]

2. Replace `Small perimeter speed` section with a merge between `Small perimeter speed` and `Very short layer time` sections like so:
- Short layer 
  - Max length (mm)
  - Max speed reduction (%)
  - Min print speed (mm/s)
- Very short layer
  - Max length (mm)
  - Max speed reduction (%)
  - Min print speed (mm/s)

I believe this way we'll be able to achieve faster and better looking prints that are easier to configure and ship better options out of the box.

supermerill, 2021-12-01:
I'll look again after the gcode rewrite as currently, that "short layer time" part of the slicer is extremely comlpicated and is really only a post-process thing after the gcode was created.

pockata, 2021-12-11:
@supermerill Thank you for taking the time to review this.

If you decide to implement this request in any form, the Filament settings for decreasing fan speeds on short layers can benefit from a conversion to being based on short perimeter lengths also.

[image]

BluPix, 2022-01-16:
Hi, could the external perimeters be omitted from the slowdown? Or ideally find a spot in the middle of the infill and wait there, or slow down only in the infill?

supermerill, 2022-01-17:
@BluPix 
That was the case long ago. proportional decrease and then the external perimeter is not enough.
But, from prusa, this create more problems than it solves, because of the extreme variation in speed.
So now they moved to a not-proportional algo: it decreases the fastest thing first, and continue until everything is at the same speed.