supermerill/SuperSlicer #1956: Error "Coordinate outside allowed range"

stuwe911, 2021-11-23:
**Describe the bug**
In version 2.3.57.5 i get this error "Coordinate outside allowed range" when trying to slice. I do not get this error with the identical project file in version 2.3.56.9. I also get the occasional application crash with the new version. Opening the file in the old one works just fine.

**To Reproduce**
Steps to reproduce the behavior:
1. Load attached project file in 2.3.57.5
2. slice it
3. Load attached project file in 2.3.56.9
4. slice it

**>> Project File <<**
[MC-24 Battery bay.zip](https://github.com/supermerill/SuperSlicer/files/7590727/MC-24.Battery.bay.zip)

**Expected behavior**
At least identical or even more stable than the old version

**Desktop (please complete the following information):**
 - OS: Windows 7
 - Version 2.3.56.9 and 2.3.57.5

supermerill, 2021-12-01:
can't reproduce on 2.3.57.6 or the nightly.

stuwe911, 2021-12-08:
well then i guess it has something to do with my printer or even PC. no biggie - i will just work with the older version. btw i had many different files with that same error... very strange. thanks for checking it out