supermerill/SuperSlicer #1957: How can I remove the (blue) overhang detection

AndyCap777, 2021-11-24:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe how it would work**
Be as descriptive as your skill level allows you.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or paint drawing about the feature request here.

AndyCap777, 2021-11-24:
The blue overhang detection is messing up my model. How can I switch it off and use normal support.

neophyl, 2021-11-24:
Please attach the requested 3mf project file zipped up. With that we can see exactly what your issue is, possibly modify the settings and post the altered one back.

AndyCap777, 2021-11-24:
I am not sure why you want me to do all that. All I am asking is how do you switch off the "Overhang Perimeter" It's shown as blue in the "Feature Type" bog in the top left corner.

neophyl, 2021-11-24:
It’s not quite that simple. First you have opened an issue, so are reporting a bug, those ask for an example project so we can see the issue.  Regardless of that you haven’t even included a picture indicating what exactly you want to switch off. 

Any answer given will be a guess as you haven’t provided needed information. I try and answer questions if I can but I dont like guessing unless it’s unavoidable. 

Dark blue is classed as overhang. Unless you mean the light blue ? That’s bridge infill. Which blue do you mean ?  Anything classed is overhang/bridging gets printed with a layer height the same as the nozzle width. This is because technically it’s not squished into the layer below so it’s treated as being a round extrudate rather than being the layer height. It’s one of the annoying things about slicer. 
If my ASSUMPTION is correct then If I had to guess then I would suggest you change the settings around overhang angle and flow %.  But it differs depending on the model.  With a project we could just see and make changes, see if the suggestion works for this instance and post a modified project back. You could then open it and see what was done to modify it. 

Basically including a project from the start streamlines the whole process which is why the issue template asks for one.

macchinette, 2021-11-24:
Set the overhangs_width_speed and overhangs_width parameters to zero.

[image]

AndyCap777, 2021-11-24:
Thanks Macchinette It worked