supermerill/SuperSlicer #1958: M600 graphic bug

wnc2, 2021-11-25:
**Describe the bug**
if you enter M600 in a "custom G-code" (right click on colorprint bar) only one layer changes color, instead of up to the end

**Expected behavior**
[image]

**Screenshots**
[image]

**Desktop (please complete the following information):**
macOs monterey 12.1

**Additional context**
Add any other context about the problem here.

supermerill, 2021-12-01:
can i have your porject file? 'file -> save porject as) (zip & drop it here)

wnc2, 2021-12-01:
Hi supermerill,

You can reproduce it with every STL:

1) add an M600 code using the colorprint dropbar on the right

THEN try to add a custom g-code on the same layer, the option doesn’t
appear anymore, so you can’t add a custom g-code if you already have set an
M600:

Obviously you can manually add a M600 command in the custom g-code, but in
this case there is a graphical bug, you see only a thin line instead of
another color for upper layers, but the M600 is correctly placed and it
works

or Android

supermerill, 2021-12-01:
Ok
that's because you used a customgcode, so it display where it is with a grey layer.
If you want to modify the colorchange gcode, modify it in the printer -> custom gcode tab