supermerill/SuperSlicer #1959: hole_size_compensation applies to rectangles and squares

macchinette, 2021-11-25:
**Describe the bug**
Hole_size_compensation applies to rectangles and squares

**To Reproduce**
Steps to reproduce the behavior:
1. Open 3mf attachment
2. Cut files
3. Scroll down to layer 177
4. See error boat cabin

**>> Project File <<**

[Hole_size_compensation.zip](https://github.com/supermerill/SuperSlicer/files/7604895/Hole_size_compensation.zip)

**Expected behavior**
Apply the function to round holes only?

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 Pro
 - Version SuperSlicer_2.3.57.6

**Additional context**
I know that using the hole_size_threshold parameter you can limit the trigger area, but if there are large holes then it also applies to smaller squares and rectangles than the holes.

supermerill, 2021-12-01:
I guess i can add the same checks than for the polyholes.

supermerill, 2021-12-01:
also, it may be useful o add a "gradient" for when the hole stop and then restart (when the vertical hole is broken open by a horizontal hole)

macchinette, 2021-12-06:
> Immagino di poter aggiungere gli stessi controlli che per i polihole.

Thanks for everything you do!

telepresencebot2, 2022-08-13:
Isn't this working as intended? It never claims to only apply to round holes.

macchinette, 2022-08-14:
The operation popup indicates the opposite of how it actually works.  Then there is the bug that applies to all closed perimeters empty in the center as if they were round holes.

Inviato da iPhone

> Il giorno 13 ago 2022, alle ore 22:51, telepresencebot2 ***@***.***> ha scritto:
> 
> Isn't this working as intended? It never claims to only apply to round holes.

telepresencebot2, 2022-08-14:
Yes, there is a separate ticket open for the tool tip being incorrect about the direction of compensation, but nothing about the tool tip or the name of the feature suggests it should only apply to round holes. It says it applies to interior perimeters.

macchinette, 2022-08-15:
okay it is not specified both for round holes, but since the function to enlarge and narrow the internal perimeters already exists and it is XY compensation, where it says Vertical Hole shrinking compensation is very useful for all round holes which are normally smaller and it doesn't happen to me for all the other closed loop perimeters that aren't round, so if it only applied to round holes it would be a wonderful thing.

Inviato da iPhone

telepresencebot2, 2022-08-15:
XY compensation applies to external perimeters. (It claims in its tool tip to only apply to exterior shells, have not tested though)
Vertical hole compensation is for internal perimeters.

macchinette, 2022-08-15:
it is not true XY compensation also applies to the internal perimeters, I'm sure of it because I tried it, for what a function only for round holes is fundamental, because all the round holes in my case are smaller, but the internal perimeters of an empty piece inside no.

Inviato da iPhone

> Il giorno 15 ago 2022, alle ore 13:22, telepresencebot2 ***@***.***> ha scritto:
> 
> XY compensation applies to external perimeters. (It claims in its tool tip to only apply to exterior shells, have not tested though)
> Vertical hole compensation is for internal perimeters.

bitingsock, 2022-08-17:
For clarification, is there currently any functional difference between "vertical hole shrinking compensation" and "XY compensation > inner"?

telepresencebot2, 2022-08-17:
"xy compensation > inner" says in it's tool tip that it applies to inner shells. "xy compensation > outer" also states that it should probably always match "xy compensation > inner".

Also, "vertical hole compensation" allows changing the threshold for what is considered a hole and what isn't, "xy compensation > outer" does not.