supermerill/SuperSlicer #1961: Issues when slicing curves/arcs compared to PrusaSlicer

Tirpitz93, 2021-11-25:
**Describe the bug**
Slicing the same object with the near identical settings results in much larger GCODE file from superslicer. This then results in failed prints very predictably as the printer basically stalls when printing curves.

**To Reproduce**
Steps to reproduce the behavior:
Slice object in super slicer (using setting imported from Prusa slicer).

**>> Project File <<**
*Please* upload a ZIP archive containing the project file used when the problem arise. Please export it just before the problem occur. Even if did nothing and there is no object, export it as it contains your current configuration.
Zip should contain both projects and GCODEs
[Prusa VS Super.zip](https://github.com/supermerill/SuperSlicer/files/7605010/Prusa.VS.Super.zip)

**Expected behavior**
I would expect similar although not identical GCODE files
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows, 
 - Version: 2.3.57.6_win64_211118 & 2.3.57.3_win64_211118
 - 
**Additional context**
Not sure if there is a setting I am missing that will allow the GCODE to be generated better, or comparably to PrusaSLicer like enabling G2/3 which AFAIK the prusa MK2 supports

Tirpitz93, 2021-11-27:
reducing processing limit to '150' only has a small impact on the file size, I am not sure how this will impact on quality.

supermerill, 2021-12-01:
Prusa has a hardcoded resolution of 0.0125
if I change the resolution from 0.002 to 0.0125, thenit output a 512kio file, lower than the prusa one.

Tirpitz93, 2021-12-03:
> Prusa has a hardcoded resolution of 0.0125 if I change the resolution from 0.002 to 0.0125, then it output a 512kio file, lower than the prusa one.
> 
> > This then results in failed prints very predictably as the printer basically stalls when printing curves.
> 
> Set a good "gcode per second" to ensure it doesn't happens anymore. Some has also some success by increasing the buffer size a bit on their firmware.

Thanks for getting back to me.
I found the printer was still struggling with 150 commands/s (the recomended typical rate for 8bit controllers).