supermerill/SuperSlicer #1962: Layer shift on stock settings

yasaslive, 2021-11-26:
**Describe the bug**
Using Super slicer stock settings and klipper print a object ended in layer shifts. But not occur when slice with prusaslicer .
 
**To Reproduce**
Steps to reproduce the behavior:
Open super slicer and place a STL file on the bed and slice. (fill gap disabled) 

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**>> Project File <<**
*Please* upload a ZIP archive containing the project file used when the problem arise. Please export it just before the problem occur. Even if did nothing and there is no object, export it as it contains your current configuration.
[TopPlate_2040_V821.1.zip](https://github.com/supermerill/SuperSlicer/files/7606052/TopPlate_2040_V821.1.zip)

**Expected behavior**
A good slicer file

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

**Desktop (please complete the following information):**
 - OS: [e.g. MacOS, Linux, Windows]
 - Version [e.g. 2.3.55.5], Use `Help->About`
Version 2.3.57
Windows 11
Processor	Intel(R) Core(TM) i7-8700K CPU @ 3.70GHz   3.70 GHz
Installed RAM	16.0 GB
System type	64-bit operating system, x64-based processor

**Additional context**
No any hardware issues and double checked the printer for faults. Restarted multiple times to check whether it was a klipper issue.

supermerill, 2021-12-01:
I'm not an hotline for printer problem.

And your zip contains only the object & gcode, no project file, so i don't have your settigns.
I think i have to change the template to add some precision on what's needed...

no lift? over-extrusion?
ask in reddit or in the facebook group of your printer.

yasaslive, 2021-12-01:
Sorry for the trouble. I tired several prints with super slicer and with prusaslicer. The prusaslicer sliced prints were okay but superslicer once were consistently shifting layers. I updated the superslicer to the latest and issue was fixed!

closed 2021-12-01 by yasaslive