supermerill/SuperSlicer #1963: Missing Prusa SL1

pandel, 2021-11-26:
Hi!

I am trying to add a SLA printer but cannot find any profile (Prusa SL1). Doesn't Superslicer support SLA at all?

Regards,
Holger

supermerill, 2021-12-01:
configuration -> wizard -> prusa sla ?

neophyl, 2021-12-01:
SLA seems to be missing from the wizard when I tried it in 2.3.57.6.  Just updated to the latest config when prompted too.

pandel, 2021-12-01:
@supermerill 
There is no category Prusa SL atm:

[image]

supermerill, 2021-12-01:
oh, right. Weird.

supermerill, 2021-12-01:
seems like I merged it a bit too aggressively. Will be fixed

If you want to fix it yourself: in the file resources/profiles/prusaresearch.ini:
replace
```
name = Prusa Research
```
by
```
name = Prusa {technology}
full_name = Prusa Research {technology} Technology Printers
technologies = FFF; SLA
```

pandel, 2021-12-01:
Tried your solution, but the printer still does not show up. But it's ok, you will come up with a solution in the next release :-), many thanks for taking care!

supermerill, 2021-12-01:
If you laready imported a prusa printer, there is a cache file in your configuration/vendor directory. Modify it or start clean.

neophyl, 2021-12-01:
Yeah, if you start Super Slicer and use Help>Show configuration folder you can find a sub folder for the vendor that Merill mentions.  In there is a copy of the prusa research.ini file.  If you change that one then run the wizard the sla option appears.  Just tried it on my work pc.

pandel, 2021-12-01:
Oha! One really has to know about this caching thing. It is working now, many thanks!

closed 2021-12-22 by supermerill