supermerill/SuperSlicer #1964: [FR] Add option for monotonic bridging pattern

kb-elmo, 2021-11-26:
**Is your feature request related to a problem? Please describe.**
Right now bridge infill is printed like standard rectiliniar infill which results in the slicer breaking up the infill into mutliple sections.
This can result in different amounts of sagging during the print due to them not being printed at once.

Example:
[image]
The bridging infill in the letters is partially printed, then the slicer proceeds to print the normal gyroid infill first before completing the bridging. This results in parts of the bridges to sag more than the rest due to different cooling times.

**Describe the solution you'd like**
Add an option to make the bridging pattern monotonic so that it is printed at once without being broken apart by the slicer.

**Describe how it would work**
Monotonic infill would cause the bridges to cool down evenly and thus reduce irregular sagging.

neophyl, 2021-11-26:
Nice idea.  Kind of like it. However monotonic is the wrong term, that just means infill is printed in the same direction, it doesn’t mean that it’s not split up into sections.

kb-elmo, 2021-11-26:
> Nice idea. Kind of like it. However monotonic is the wrong term, that just means infill is printed in the same direction, it doesn’t mean that it’s not split up into sections.

ah ok I just thought that this was what it meant haha :D

bwnance, 2022-07-12:
@supermerill ok! thanks

DanielSmith1239, 2023-01-19:
Any updates on this?

bwnance, 2023-01-19:
@DanielSmith1239 i ended up writing this myself! ([PR here](https://github.com/supermerill/SuperSlicer/pull/2997)) the changes were merged and are in the Arachne 2.5 release - https://github.com/supermerill/SuperSlicer/releases/tag/2.5.59.0

bwnance, 2023-01-19:
@supermerill good to close this issue due to the merge of #2997

DanielSmith1239, 2023-01-19:
Ah ok, so it's not in 2.4 then? My issue is similar, where SuperSlicer's bridging pattern isn't continuous (see pictures). Is that what you fixed?

PrusaSlicer:
[image]

SuperSlicer:
[image]

DanielSmith1239, 2023-01-19:
Also note that PS has the normal infill before bridging, and SS has the opposite. Might have an impact since it'll be missing anchors.

DanielSmith1239, 2023-01-19:
Update: looks like your update fixed my bridging issue, thanks! But now I have to figure out how to get the infill to print before the bridges, so I'd really appreciate any input you have on that

bwnance, 2023-01-20:
@DanielSmith1239 not totally sure on why susli prints sparse infill after bridged - i'd assume that would be an easy addition to the codebase, if i have some time in the next couple weeks i'll take a swing at it.

closed 2024-05-03 by supermerill