supermerill/SuperSlicer #1965: Hello i cant have a preview gcode with superslicer why ?

Kounous972, 2021-11-26:
Hello use a scripts for superslicer but cant have a preview its working with prusaslicer and cura but not working for superslicer ?
[image]

neophyl, 2021-11-27:
Merrill will need a saved zipped up project file from super slicer and possible a copy of your script to have any chance of figuring out what is happening. Can you please include them.

Pavulon87, 2021-11-29:
It's not a problem with SuSi, but with the "CreateHeaderAndThumbnail.py" script or with the "Post-processing scripts" setting.
Open [Command Prompt](https://www.howtogeek.com/235101/10-ways-to-open-the-command-prompt-in-windows-10/) and try to run:
`C:\Python\python.exe "C:\Users\kouno\AppData\Roaming\SuperSlicer\scripts\CreateHeaderAndThumbnail.py" "E:\piggybank_close_v5_36m_0.16mm_205C_PLA_ENDER3V2.gcode"`
Or remove this line from Post-processing scripts

Kounous972, 2021-11-29:
> It's not a problem with SuSi, but with the "CreateHeaderAndThumbnail.py" script or with the "Post-processing scripts" setting. Open [Command [...]

Hi on prusaslicer its workly perfect but when try it on SS is not working

supermerill, 2021-12-01:
> Hi on prusaslicer its workly perfect but when try it on SS is not working

that means your python script encounter something he doesn't like in the SuSi gcode and crashed.