supermerill/SuperSlicer #1966: Bug: With a RAFT, SS uses "bridge infill" where it should be "Solid infill" on objects 1st layer

eduncan911, 2021-11-26:
**Describe the bug**
First let me say, what an awesome slicer!  Thank you!

My normal 1st layers are perfect.  And even the RAFT's 1st layer is great.  However, the model's "1st layer" is not using Solid Infill; but instead, it is using "Bridge Infill".

Also: "_Only one parameter: On First Layer_" does not apply when using RAFTs: it still prints all 3 parameters on the object's 1st layer.  I don't think this applies to this bug though, because I can uncheck it, and it still prints bridge infill only.

**To Reproduce**

1. Add an object, Slice.  You should see a normal first layer (in this screenshot, I am using 1 parameter for first layer, but that doesn't matter)

[image]

2. Now, Add a RAFT (Print Settings -> Support material -> Raft settings (I uses 2 raft layers)
3. Slice again, and view layer 3 (first layer of object).  It's using Bridge infill instead of Solid infill.  See screen below:

[image]

**>> Project File <<**

[SuperSlicer_configs.zip](https://github.com/supermerill/SuperSlicer/files/7610886/SuperSlicer_configs.zip)

**Expected behavior**

Slicer produces gcode for a Solid infill on object's first layer.

**Desktop (please complete the following information):**
- OS: Pop_OS 21.04 (aka Ubuntu 21.04)

**Additional context**

Using Ellis' PIF SS configuration, as of Nov 21.

I have been getting great 1st layer results, even with "Only one parameter: On first layer"  I first thought that was the issue.

However, unchecking "On first layer" doesn't change it.  SS still slices in "bridge infill" where it should be "Solid infill"

eduncan911, 2021-11-27:
Looks like this is a problem with Supports as well (though technically, SS/PS treats Rafts as Support, yes).  

I was adding support for a different object today, and the same thing is happening on the object's first layer.

[image]

I can't seem to find a configuration setting that would affect this.

neophyl, 2021-11-28:
2 things.
First eduncan is correct in that Super Slicer/Prusa Slicer treat anything printed on supports (that includes rafts) as being bridged.  That means that it uses bridging flow and assumes the layer height is the same as your nozzle width.  Wether that is right or not its the way its always been (at least up until PS2.4 where they have a new option to disable thick bridges).   One reason I'm looking forward to the future merge of PS 2.4 into SuSi.
Basically rafts are pointless and if at all possible its better to sort out your bed adhesion without relying on rafts.

2nd thing, and this is the important one.  Bridging was redone in 2.3.57.6.  Under Print Settings>Width & Flow>Overlap there are now settings for Bridge Lines density.  They are expert level settings so you must be in expert mode.  Your profile has both min and max set to 50%.  That's why its doing a wide spacing of the bridge lines.  If you set them to 100% on both then you will get a more expected result.

Why your profile has them set to 50% I don't know though.

supermerill, 2021-12-01:
If you have soluble support, you have to set your supports to soluble mode.
Then, it won't use bridges for printing on top of support, as it can squish the extrusion over the soluble surface.

eduncan911, 2022-01-26:
thanks guys!  note though that in the 2nd photo above, those are supports - not a raft (though kind of the same thing).

it's such a weird thing that prevents basically any type of support or raft.  like, what's the point if the bottom-most layer is always going to be a bridge, and not a normal bottom layer.

closing as it seems to be by design of PS/Slic3r and not something for SS.

closed 2022-01-26 by eduncan911