supermerill/SuperSlicer #1969: Crash when slicing

Psyc0Flyer, 2021-11-28:
**Describe the bug**
I load my file for bed wheels to slice on my Ender 3 v2 and it crashes after slice saying not enough ram but have 32GB ram installed

**To Reproduce**
Steps to reproduce the behavior:
1. Load file
2. Slice

**>> Project File <<**
[File is too large for github?](https://drive.google.com/file/d/1hxuG4qLsBnqA-eLTaiG97r6fwFMzBuie/view?usp=sharing)

**Expected behavior**
SS crash

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
[image]

neophyl, 2021-11-28:
Your file is the amf.  I couldn't get the same version of Super Slicer to even load that amf.  Says no elements found.  

Generally what is needed is your saved project file from Super Slicer.  So File>Save Project As just before you click slice.  The project file (.3mf) will contain all your profile settings in use as well as the model.

Psyc0Flyer, 2021-11-28:
> Your file is the amf. I couldn't get the same version of Super Slicer to even load that amf. Says no elements found.
>
> Generally what is needed is your saved project file from Super Slicer. So [...]

Ok, sorry. I have updated the link.

neophyl, 2021-11-28:
I'm trying to load it into the latest 2.3.57.6 and I'm getting an error on trying to load the project file "Error (unclosed token) while parsing '3D/3dmodel.model' at line 36515977"

I cant figure out why 4 wheels is such a large file anyway, unless those wheels are at some insane resolution that a fdm cant print at anyway.  When extracted that model is over 2 gig so I'm not surprised it wont slice.  Where did you get the original source from ?

Psyc0Flyer, 2021-11-28:
> I'm trying to load it into the latest 2.3.57.6 and I'm getting an error on trying to load the project file "Error (unclosed token) while parsing '3D/3dmodel.model' at line 36515977"
> 
> I cant figure out why 4 wheels is such a large file anyway, unless those wheels are at some insane resolution that a fdm cant print at anyway. When extracted that model is over 2 gig so I'm not surprised it wont slice. Where did you get the original source from ?
> 
> Edit - also microsoft 3d builder cant open the 3mf either.

Geesh, ok. I got them from [here](https://www.thingiverse.com/thing:4743835/files) and maybe my fusion parameters are set too low for output? I don't remember messing with them but not sure what to set the deviation sliders to?
[image]

neophyl, 2021-11-29:
I dont use fusion so don’t know what those sliders control. The fusion forums should be able to tell you what is suitable for stl or other output formats.