supermerill/SuperSlicer #1970: Artillery Genius Pro stops during printing.

J0hanss0n, 2021-11-28:
**Describe the bug**
Artillery Genius Pro stops during printing of attached project file. This also happens when the same structure is part of larger models.

**To Reproduce**
Open attached project. Slice. Export. Print.
It always stops after 2 to 3 layers.

**>> Project File <<**
[bolt_superslicer.zip](https://github.com/supermerill/SuperSlicer/files/7613164/bolt_superslicer.zip)

**Desktop**
 - OS: Windows 10
 - Version 2.3.57.6

**Additional context**
PrusaSlicer works in this case.
[bolt_prusa.zip](https://github.com/supermerill/SuperSlicer/files/7613172/bolt_prusa.zip)

What I can see, is that the SuperSlicer Gcode file is much bigger than the PrusaSlicer file.

Regards, Johannes

J0hanss0n, 2021-11-28:
Update:
Even if I reduce the Processing limit to 150 G1/s the GCode file stays big (> 2Megabyte). Because of still over 2000 "wipe and retract" lines per retract. (e.g. G1 X108.291 Y110.708 E-0.02046 ; wipe and retract)
I think the large amount of wipe and retract commands generated are clogging the queue.

J0hanss0n, 2021-11-28:
Seems related to #1961
Sorry for opening a more or less duplicate.

J0hanss0n, 2021-11-29:
I reviewed the code a little, and from my extremely limited point of view, the reason for the high amount of "wipe and retract" commands despite limited Gcode per second is lines 3304 and 3305 of GCode.cpp:

```    
if (m_wipe.enable)
        m_wipe.path = paths.front().polyline;  // TODO: don't limit wipe to last path
```

The simplified wipe path has already been set in the call of extrude_path (line 3298), where the max_gcode_per_second value had been applied. The above lines afterwards assign a not simplified polyline.

Regards, Johannes

supermerill, 2021-12-01:
can't see a difference in retractions between prusa & SuSi

[image]

I can see teh problem with the not-simplified wipe path, though.

J0hanss0n, 2021-12-01:
Thanks for the reply. 
The main difference is the amount of instructions. Its over 2000 Gcodes per retract (Retraction is under 2mm). With a retraction speed of 40mm this estimates to about 40000 Gcode/s. This stops my printer during execution.

supermerill, 2021-12-01:
reduce the resolution, or put a lower "max G1 per second"

resolution: 
 * prusa: 0.0125 (hardcoded) 
 * Susi: default to 0.002

a max G1 per second of 150 should be low enough for most printers.

>  This stops my printer during execution.

Can you test the higher value possible for 'maximum G1 per second' , so i can modify the included profile?

J0hanss0n, 2021-12-01:
I will try to lower the resolution once I'm back home, but as already mentioned lowering the "maximum G1 per second" does not work because of the not simplified wipe path where this value is not applied. It's the retract that causes the printer to stall:
Lowering to 150 leads to about 100 to 200 Perimeter and Gapfill commands, followed by more than 2000 to 3000 Wipe and retract commands per Layer.